espruino/BangleApps #3033: E.showMenu() fails with special character on firmware 2v19

nxdefiant, 2023-09-27:
The following snippet with should show a menu entry with a german umlaut
```
const mainMenu = {};

mainMenu[JSON.parse('"Hello \u00E4"')] = undefined;
E.showMenu(mainMenu);
```
worked with firmware 2v18:
[image]

But fails with firmware 2v19:
> Uncaught Error: Expecting first argument to a valid Image
 at line 2 col 337
...r[a];k=g.wrapString(a,b.w-l);1<k.length&&(k=g.setFont("6x15"...
                              ^
in function "draw" called from line 1 col 142
...,{x:a.x,y:m(b),w:a.w,h:d.h});g.setClipRect(0,0,g.getWidth()-...
                              ^
in function "draw" called from line 3 col 339
...w==n},f=e.scroll&-2;e.draw();g.flip();return e
                              ^
in function called from line 1 col 28
p.scroller=E.showScroller(u)
                           ^
at line 1 col 28
p.scroller=E.showScroller(u)
                           ^
in function "t" called from line 8 col 247
....scroller.drawItem(a)}}};t();return p
                              ^
in function called from line 1 col 20
E.showMenu(mainMenu);
[image]

thyttan, 2023-09-27:
Possibly related: I haven't tried to pin it down, but semi-recently messages started showing with missing sections for me. I enabled "Text as Bitmaps" in Gadgetbridge and that seemed to fix my issue. But I'm almost certain I didn't need this setting toggled on to receive complete messages before.

gfwilliams, 2023-09-28:
Ok, thanks - if you need to work around it, this works fine: `E.showMenu({"Hello \xE4": undefined});`

How did this one come up? I thought usually we convert characters that are in the UTF8 code range into the `\x..` form now.

The issue appears to be to do with how UTF8 strings are stored when they're object keys - for compatibility we treat `\xE4` as a non-UTF8 string.

```
>trace({'Hello \u00e4': undefined})
#1403[r0,l1] Object {
  #1324[r1,l2] Name String [2 blocks] "\0\0\0\0o \xC3\xA4"    undefined
}
```

This one could end up being a bit of a pain to sort out since the way we tag strings as being UTF8 isn't really compatible with this. I'll have to have a think how we work around it.

And yes, since we added UTF8 support in Bangle.js, we went through a stage where Gadgetbridge was still sending UTF8 chars (like 0xE4) but Bangle.js had to interpret them as UTF8 (but it was invalid) - but I thought we'd got that fixed now - Gadgetbridge shouldn't be sending those chars and also Bangle.js should treat invalid unicode as normal non-UTF8 data

gfwilliams, 2023-09-28:
Just a note that this is hopefully fixed with https://github.com/espruino/Espruino/commit/dde4d3a5ca52782fbe7cefacb18d52d9655b485a now - but I'll need to do a bit more testing

nxdefiant, 2023-09-28:
Cool thx, I'll fetch the cutting edge build.

The original bug was seen in [noteify](https://banglejs.com/apps/?id=noteify), a text with a special character was entered from the webinterface, save as json on the device and shown in the app.

gfwilliams, 2023-09-29:
Ok, just checked on a Bangle and this now all seems work work fine - phew!

UTF8 always seems to end up being a bit of a nightmare, and we're not really seeing any of the benefits at the moment. Hopefully soon though :)

closed 2023-09-29 by gfwilliams