espruino/BangleApps #3036: [weather&homeassistant] Some apps no longer shows clock_info

kamilkrz, 2023-10-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Hi,
Weather (with owm provider but I think that is not as important) and homeassistant app stopped showing clock_info.

In contrast I also use "clock info module", "calendar clockinfo" and also installed "firmware clockinfo" for testing and those working like a charm.

Tested with lato and  lcd clock plus watchfaces.
I believe all mentioned above worked in early 2v18 (beta) but I am not fully sure as I only used watchface with one clock_info space and most of the time was stuck with steps.
I remember that homeassistant was a bit buggy with icons bigger than reserved space for clock_info widget in many of watchfaces but that was minor issue.

bobrippling, 2023-11-12:
Hey - can you share which versions of the homeassistant and weather apps you're using? If you install another clock info app, does it show up?

kamilkrz, 2024-09-27:
I will readdress similar issue as this was part of me testing things - reinstall apps + bootloader helps then and I totally forgot about updating it here.

closed 2024-09-27 by kamilkrz