espruino/BangleApps #3043: [Pebble Clock with distance] clock showing wrong time / frozen

pavelmachek, 2023-10-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

I looked at my watch and time was off by 30 (or so) minutes. I thought something went wrong with system time, but after starting another clock, they showed correct time, and going back to "Pebble with distance" showed correct time.

I suspect clock just froze and did not show correct time for 30? minutes? I was charging my watch in relevant timeframe.

Not sure how to debug this, it only happened once so far.

### Installed apps

notify (0.13), sched (0.22), widbt (0.09), widlock (0.08), widid (0.03), sixths (0.01), run (0.16), pebbled (0.04), mylocation (0.10), widclk (0.07), boot (0.59), health (0.27), setting (0.63), recorder (0.30), bikespeedo (0.06), clock_info (0.07), widadjust (0.02), astrocalc (0.06), orloj (0.01), altimeter (0.03), widbat (0.12), f9lander (0.03), a_clock_timer (1.00), folderlaunch (0.03), alarm (0.43), gipy (0.22), dragboard (0.09), waypoints (0.01), magnav (0.06), waypointer (0.09), waypoint_editor (0.02), spacew (0.02)

gfwilliams, 2023-10-13:
I imagine what happened is at some point during drawing, something had an error, and then it didn't manage to get to schedule the next draw.

You could just move the 'queue next draw' code to the top of the draw function, which would probably fix this: https://github.com/espruino/BangleApps/blob/master/apps/pebbled/pebbled.app.js#L105C3-L110

But if you want to find the actual issue I'd enable the debug log in settings, then if it happens again you can look at the debug log and find out what the stack trace was.

**However:** I'm really anxious about the BangeApps repo being full of a bunch of "This happened once and I can't reproduce" issues, as I can just see them piling up and nobody ever bothering to follow up and close if they never happen again... For instance when do you plan to close this issue if you never see the issue happen again?

pavelmachek, 2023-10-13:
Well, I consider showing stale time pretty serious. I guess moving the draw would make sense.

I'll take a look at the debug logging.

Would it make sense to have some kind of watchdog (if clock have not repainted for 2 minutes, display failure screen?) or if we should be killing applications that produce stack traces so we don't have this kind of silent fails?

gfwilliams, 2023-10-13:
I think that's a hard one to do - we don't know which timeout is for a redraw or something else.

Someone could make an app that caught any uncaught exceptions and restarted I guess. But my point is you've seen this happen once, with one app. It's not exactly a common occurrence.

When do you plan to close this issue if you never see the issue happen again?

Bugs that happen one time on a watch that I have no real idea of the configuration of, in code that I didn't write, with no stack trace and which are not reproducible are basically impossible for me to reliably pinpoint and fix.

pavelmachek, 2023-10-28:
I understand that your situation is not easy. OTOH showing wrong time on watch is quite critical failure and.... well... bugs should be reported.

bobrippling, 2023-10-30:
I've added the suggested fix - @pavelmachek, if you do see the issue again and have debug logs turned on, if you could pop them here that'd be great, thanks.

PR #3064 (2023-10-30): pebbled: prevent exceptions halting the draw cycle
This ensures `pebbled` continues to draw - similar changes could be made to `pebble` and `pebblepp`, but the former has had no bugs reported and the latter's code is simple enough that it doesn't appear likely to raise exceptions when drawing.

The issue is likely an exception in `pebbled`'s drawing code, but this solves the watch redraw problem while we wait to hear about possible debug logs.

Closes #3043.
closed 2023-10-30 by gfwilliams