espruino/BangleApps #3044: Run app malfunctioning

vkrish84, 2023-10-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

When using the Run app, it does not show the distance that I've traveled nor does it show the pace (average or current).

Also, when I press stop, to stop the timer, and start again it will ask if I want to resume the run. Regardless of whether I say yes or no, the app will start up again and it does not load properly. The only thing that is displayed is the time  (see screenshot)

[image]

Is there a fix for this? Thanks

### Installed apps

launch (0.20)
about (0.14)
widlock (0.08)
sched (0.22)
sched (0.22)
widalarm (0.01)
setting (0.63)
health (0.27)
notify (0.13)
widbt (0.09)
boot (0.59)
clock_info (0.07)
pebblepp (0.04)
hrm (0.11)
widpedom (0.22)
widbat (0.12)
alarm (0.43)
run (0.16)

gfwilliams, 2023-10-13:
Hi - I've just tried here on 2v19 with a fresh install and it seems to work fine for me.

Do you think there is anything set up in your app loader I should be aware of? Like `Minify apps before upload` or translations?

Please could you go to the Web IDE (https://www.espruino.com/ide/), click connect (top left) then `Web Bluetooth` and choose your device. Then start the run app and do the stuff you were doing, and see if anything gets printed on the left-hand side?

If it does, please could you copy it and paste it up here? It might give a hint as to what is broken

gfwilliams, 2023-10-13:
Sorry, I take that back - after the menu about resuming, I do see the blank screen, and I'll get a fix in for that.

>  it does not show the distance that I've traveled nor does it show the pace (average or current).

In the bottom left, is the GPS marker green, or red? Without GPS the app can't work out how far you've gone, so that could be what you're seeing?

I'd advise using the `Assisted GPS` app to upload AGPS data to the GPS to allow it to get a GPS lock much quicker (hopefully 30-ish seconds) - but you'll have to be outside for that

vkrish84, 2023-10-13:
> Sorry, I take that back - after the menu about resuming, I do see the blank screen, and I'll get a fix in for that.
> 
> >

Ok great thanks! and FYI it was happening for the Run+ app as well. And not sure if this is happening for you, but the clock is not restarting after I exit the app.

> In the bottom left, is the GPS marker green, or red? Without GPS the app can't work out how far you've gone, so that could be what you're seeing?
>
> I'd advise using the `Assisted GPS` app to upload [...]

Ok, will give it a try.

gfwilliams, 2023-10-13:
Yes - I fixed both of them ;)

How were you exiting the app? If I long-press the button to exit, it seems to work fine for me. Shows 'Loading...' and then the clock screen

gfwilliams, 2023-10-13:
You definitely updated to the latest version on the development app loader? https://espruino.github.io/BangleApps/

the changes won't be live on banglejs.com/apps yet

vkrish84, 2023-10-13:
Ah ok, for some reason I wasn't seeing the update. Blank screen is gone now :) Will try out the GPS later on.

Appreciate it!

medwriter01, 2023-10-14:
Hi, I am not able to get the GPS working. Even with the Assisted GPS app there doesn't seem to be a connection. I was outside for about 20 minutes and the GPS marker stayed red.

Is there perhaps something I need to change in the settings? Thanks

Darin755, 2023-10-14:
The problem is fixed by installing the "recorder" app. I don't know why but as soon as I installed it as well the problem stopped happening

medwriter01, 2023-10-15:
Are you using the reocorder app, or just installing it solved the problem?

Darin755, 2023-10-15:
> Are you using the reocorder app, or just installing it solved the problem?

Just installing it fixes the problem

mdright01, 2023-10-16:
@gfwilliams The screen works okay. I am unable to get the distance to show up when using the app. I will check with other GPS apps.

Could it be that the recorder app is triggering the GPS to run in the background?

gfwilliams, 2023-10-16:
Well, if you're recording the recorder app would turn GPS on - but the run app should do this anyway, so I'm not convinced that would be the issue.

mdright01, 2023-10-16:
> Well, if you're recording the recorder app would turn GPS on - but the run app should do this anyway, so I'm not convinced that would be the issue.

Ok. I downloaded the GPS info app, it doesn't seem like its getting a fix.