espruino/BangleApps #3048: [calendar] Swipe actions seem wrong in calendar app

jt-nti, 2023-10-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

The current calendar swipe actions seem like they're the wrong way round to me...

- Swipe left to go to the previous month
- Swipe right to go to the next month
- Swipe up (Bangle.js 2 only) to go to the previous year
- Swipe down (Bangle.js 2 only) to go to the next year

Swiping should be a drag gesture, for example how the banglejs menu works, dragging up to view items further down the menu.

Would it be ok to submit a PR to reverse the current swipe gestures in the calendar?

nxdefiant, 2023-10-15:
I'm the perpetrator that changed the calendar handling from tapping on the left/right side to swipe events.
Unfortunately I do not understand your problem, but I gather that you expect for example a swipe from left to right to go to the previous month instead of the next month?

I think the best solution is to add an option for that in the settings.

Sure PRs are welcome, but in this case I would like you to wait a few days because I'm also working on an update and I would hate to run into a merge conflict. (Or you do the change on my [branch](https://github.com/nxdefiant/BangleApps/tree/calendar)?)

jt-nti, 2023-10-15:
Swiping is definitely a nice improvement to the old tapping left and right, I just always swipe the wrong way! I wondered whether an option to switch the direction round might be ...well, an option :)

It would likely take me longer than a few days to getting round to working on it anyway, so there shouldn't be any danger of conflicts. It's probably worth keeping the issue open for a bit to get feedback on what the option might be called, and which direction would be most popular for the default.

gfwilliams, 2023-10-16:
Yes, I think this is probably just a personal preference thing? I think if someone really cares they can make a PR to add an option for it I guess? But if we change it around now we'll just get **another** issue later complaining it's the wrong way and swipe left should to go to the previous month.

... and I'm not even joking. It's happened before for other similar things.

jt-nti, 2023-10-16:
That related thread was definitely interesting. It would certainly be nice if things were consistent so rather than a calendar specific option, would a system-wide option be a possibility?

jt-nti, 2023-10-16:
No idea when I'll get a chance but if no one beats me to a PR I'll give it a go sometime.

nxdefiant, 2023-10-16:
You could try something similar to this as swipeinvert.boot.js:
```
{
  const setURIOrig = Bangle.setUI;
  Bangle.setUI = (mode, callback) => {
    if (typeof mode === "object" && mode.swipe) {
      const origSwipeCb = mode.swipe;
      mode.swipe = (dirLR, dirUD) => origSwipeCb(dirLR*-1, dirUD*-1);
    }
    return setURIOrig(mode, callback);
  };
}
```

PR #3217 (2024-02-28): swipeinv: new app
Attempts to fix #3048 as an alternative to #3202. Up for discussion.

Inverts swipe direction globally or per app, see settings. If global inversion is enabled, you can unselect the inversion per app and vice versa.

Ping @jt-nti @thyttan
closed 2024-02-28 by bobrippling