espruino/BangleApps #3050: [gassist] FYI this also works for samsung's bixby but not yet chatgpt

ccchan234, 2023-10-16:
hi,
originally i wanna modify this for chatgpt,
however i found that this app simply calls android.intent.voice_command.
i.e. it's non specifc, it calls any default voice command input.

if it is google assistant, then this app calls it.
if it is samsung bixby (as in my samsung a33), then this app calls bixby.

as in the espruino forum's project page, i wanna make it for chatgpt but currently need extra help from OPENAI as the chatgpt apk's use of intent is not well documented.

thanks

closed 2023-10-16 by ccchan234

gfwilliams, 2023-10-16:
Please can you avoid posting 'project' posts/threads like this? If you have questions please ask them in the forum - but it looks like you might already be asking for help there on exactly this?

https://forum.espruino.com/conversations/391050/#comment17161344

gfwilliams, 2023-10-16:
Well that's fine as you closed it, but posting here won't notify the author.

You'll have to look at the commit history for the app and then reference the author directly with `@` and their name.