espruino/BangleApps #3052: [cards] Widget bar obscuring QR codes

m-p-3, 2023-10-16:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

### Describe the bug
When the watch goes into "lock" mode, the widget bar shows up shortly after, obscuring the top of the QR code and making it unscannable because the timing blocks are the top are cropped.

### Expected behavior
Make sure the widget bar doesn't show up when a QR code is displayed.

### Steps to reproduce
1. Create a test QR code in Catima
2. Upload card
3. Display card
4. Wait for BangleJS to lock
5. [image]

### Installed apps

`messageicons (0.05), widalarm (0.01), authentiwatch (0.07), files (0.08), widbt_notify (0.17), calclock (0.06), widclk (0.07), info (0.03), widlock (0.08), widslimbat (0.01), fileman (0.03), weather (0.25), sched (0.22), widmessages (0.05), powermanager (0.10), drained (0.03), owmweather (0.03), mylocation (0.10), agenda (0.14), quicklaunch (0.15), locale (0.17), qmsched (0.10), wid_edit (0.05), launch (0.20), boot (0.59), android (0.32), messagegui (0.74), messages (0.62), cards (0.01), alarmqm (0.01), setting (0.64), alarm (0.44)`

nxdefiant, 2023-10-16:
@glemco

Does the QR code usually paint on the full area? If so it looks like we shoud either
- remove load/drawWidgets() calls
-  Switch to widget_utils hide/swipeOn();

glemco, 2023-10-19:
@nxdefiant  yes it does. I could make it smaller but the resolution is already not so great so I didn't go down that path..
As soon as I have some time I'll look into it. I'm not sure how it works but since it should be possible to go back and forth from the app (with widgets) to the code (without), would removing the call to drawWidgets work?

nxdefiant, 2023-10-19:
If you really want to show widgets try http://www.espruino.com/Bangle.js+Hideable+Widgets

glemco, 2023-10-20:
It would be much nicer if I could unload and reload widgets, the code for QR is already having issues as it is, saving some memory might give ti some breath.
I'll give it a go, thanks!

gfwilliams, 2023-10-20:
Yes, I'm afraid widget unloading just isn't possible, so probably the drag-down is the way to go.

For the QR code stuff, it's very cool to have it running on the watch, but did you ever consider just saving a bitmap like we do for https://banglejs.com/apps/?id=qrcode ? It'd be *way* faster and I guess ultimately maybe even a bit more flexible?

glemco, 2023-10-20:
@gfwilliams well, then I'll go for that, thanks.

> but did you ever consider just saving a bitmap

Yes definitely, just that wasn't the way it was implemented in gadgetbridge, it would be ideal if it could supply a bitmap (either generated there or by catima), but I don't think it's currently possible.

This is a proof of concept that what is made for ZeppOS can work also here (it's also not that functional with most of the codes, by the way). Unfortunately, I don't have the time not much the knowledge to implement bitmap generation from android

gfwilliams, 2023-10-20:
Ok, no worries. Just so you know, if you *can* can a `Bitmap`/drawable somehow in Gadgetbridge then in the BangleDeviceSupport class there's a function that'll turn that drawable into an Espruino-style bitmap though.

It's something I've been considering doing for the HTTP request (allowing Bangle.js to request an image and have it converted)

closed 2023-11-03 by bobrippling