espruino/BangleApps #3055: trying to set Bangle 2 to poll hrm at 100Hz via setOptions not working

jabituyaben, 2023-10-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

I'm using 

Bangle.setOptions( {hrmPollInterval :10 });

to try change the hrm sample rate to 100Hz but it only changes to 50. I also don't think this works either for turning off green-light adjust:

Bangle.setOptions( {hrmGreenAdjust :false });

Is there another options or something that needs to be done before or after these? I've tried changing power save modes and running these before and after turning on the sensor but it doesn't change anything.

jabituyaben, 2023-10-19:
Also to clarify, the option itself changes to 100 in the object properties - but by looking at recorded data, you can see it's not 100Hz, it' 50. So something related to the firmware I guess - unless, like I said, there's some other commands I need to run before or after these.

closed 2023-10-19 by jabituyaben

gfwilliams, 2023-10-20:
Hi - why did you close this? Did you find the answer?

I think there could be potential issues where the sensor needs a certain time to take a reading, so you might need to poke some registers.

But if you check out the flash counter app at https://espruino.github.io/BangleApps/?id=flashcount that manages to get around 200Hz

jabituyaben, 2023-10-20:
> Hi - why did you close this? Did you find the answer?
>
> I think there could be potential issues where the sensor needs a certain time to take a reading, so you might need to poke some registers.
>
> But [...]

I thought I added a comment here before I closed! Basically I retried setting the poll option before turning the sensor on and I got it working. It's not quite exactly 100Hz in the end but definitely changes...It's better to stick with 25Hz for the hrm I'm finding. For the LED adjustment, if I turn that off at the start I get no readings, it would be better if that option was tied to a fixed LED current that you could set which locks the adjustment as well as directly set what level you want rather than a boolean I think.

gfwilliams, 2023-10-20:
Ahh, ok - right - yes, I think you need to set the poll interval before starting the HRM, and then maybe lock the LED adjustment *after* it's had time to adjust correctly.

> it would be better if that option was tied to a fixed LED current that you could set which locks the adjustment

Someone could make a PR for this I guess - but you can always set it by poking the registers directly if you need to