espruino/BangleApps #3062: [module/suncalc.js] Calling functions multiple times leads to errors.

g-rden, 2023-10-28:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Calling functions from suncalc.js a few times creates errors. When sending following program to Bangle 2 via the web ide an error occurs. This does not happen with the emulator. Don't have a Bangle 1 for testing. Hope someone else can reproduce this.

Program:
```
var i;

for (i = 0; i < 300; i++) {
  print(i);
  print(require("suncalc").getTimes(new Date(), 0, 0, 0).sunrise);
}
```

Terminal output:
```
No errors in suncalc. Minified 10476 bytes to 4792 bytes.
>BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v19 (c) 2021 G.Williams
>
0
Date: Sat Oct 28 2023 07:41:35 GMT+0200
1
Date: Sat Oct 28 2023 07:41:35 GMT+0200
2
Date: Sat Oct 28 2023 07:41:35 GMT+0200
3
Date: Sat Oct 28 2023 07:41:35 GMT+0200
4
Date: Sat Oct 28 2023 07:41:35 GMT+0200
5
Date: Sat Oct 28 2023 07:41:35 GMT+0200
6
Date: Sat Oct 28 2023 07:41:35 GMT+0200
7
Date: Sat Oct 28 2023 07:41:35 GMT+0200
8
Date: Sat Oct 28 2023 07:41:35 GMT+0200
9
Date: Sat Oct 28 2023 07:41:35 GMT+0200
10
Date: Sat Oct 28 2023 07:41:35 GMT+0200
11
Date: Sat Oct 28 2023 07:41:35 GMT+0200
12
Date: Sat Oct 28 2023 07:41:35 GMT+0200
13
Date: Sat Oct 28 2023 07:41:35 GMT+0200
14
Date: Sat Oct 28 2023 07:41:35 GMT+0200
15
Date: Sat Oct 28 2023 07:41:35 GMT+0200
16
Date: Sat Oct 28 2023 07:41:35 GMT+0200
17
Date: Sat Oct 28 2023 07:41:35 GMT+0200
18
Date: Sat Oct 28 2023 07:41:35 GMT+0200
19
Date: Sat Oct 28 2023 07:41:35 GMT+0200
20
Date: Sat Oct 28 2023 07:41:35 GMT+0200
21
Date: Sat Oct 28 2023 07:41:35 GMT+0200
22
Date: Sat Oct 28 2023 07:41:35 GMT+0200
23
Date: Sat Oct 28 2023 07:41:35 GMT+0200
24
Date: Sat Oct 28 2023 07:41:35 GMT+0200
25
Date: Sat Oct 28 2023 07:41:35 GMT+0200
26
Date: Sat Oct 28 2023 07:41:35 GMT+0200
27
Date: Sat Oct 28 2023 07:41:35 GMT+0200
28
Date: Sat Oct 28 2023 07:41:35 GMT+0200
29
Prompt not detected - upload failed. Trying to recover...
Date: Sat Oct 28 2023 07:41:35 GMT+0200
30
Date: Sat Oct 28 2023 07:41:35 GMT+0200
31
Date: Sat Oct 28 2023 07:41:35 GMT+0200
32
Date: Sat Oct 28 2023 07:41:35 GMT+0200
33
Date: Sat Oct 28 2023 07:41:35 GMT+0200
34
Date: Sat Oct 28 2023 07:41:35 GMT+0200
35
Date: Sat Oct 28 2023 07:41:35 GMT+0200
36
Date: Sat Oct 28 2023 07:41:35 GMT+0200
37
Execution Interrupted
at line 1 col 69
...,lw,n);return solarTransitJ(a,M,L)
                                    ^
in function "getSetJ" called from line 1 col 410
...getSetJ(h0,lw,phi,dec,n,M,L),Jrise=Jnoon-(Jset-Jnoon),result...
                              ^
in function "getTimes" called from line 3 col 56
...etTimes(new Date(), 0, 0, 0).sunrise);
                              ^
>echo(1)
=undefined
>
{"t":"act","stp":0,"hrm":0,"mov":100}
```

Console output:
```
>>> Loading modules
loadModule(suncalc)
ERROR: getURL("https://www.espruino.com/modules/suncalc.min.js") error : HTTP 404
ERROR: getURL("https://www.espruino.com/modules/suncalc.js") error : HTTP 404
ERROR: getURL("https://banglejs.com/apps/modules/suncalc.min.js") error [object ProgressEvent]
 - suncalc requires []
>>> Minifyingn suncalc
[notify_info] No errors in suncalc. Minified 10476 bytes to 4792 bytes.
>>> Minification complete
>>> Minifying
>>> Minification complete
Found a prompt... great!
>>> Sending...
---> "\u0010reset();\n\u0010print()\n\u0010setTime(1698526760.143);E.setTimeZone(2)\n\u0010Modules.addCached(\"suncalc\",\"!function(PI,sin,cos,tan,asin,atan,acos,rad,dayMs,J1970,J2000,e,SunCalc,times,J0){'use strict';function toJulian(date){return date.valueOf()/dayMs-.5+J1970}function fromJulian(j){return new Date((j+.5-J1970)*dayMs)}function toDays(date){return toJulian(date)-J2000}function rightAscension(l,b){return atan(sin(l)*cos(e)-tan(b)*sin(e),cos(l))}function declination(l,b){return asin(sin(b)*cos(e)+cos(b)*sin(e)*sin(l))}function azimuth(H,phi,dec){return atan(sin(H),cos(H)*sin(phi)-tan(dec)*cos(phi))}function altitude(H,phi,dec){return asin(sin(phi)*sin(dec)+cos(phi)*cos(dec)*cos(H))}function siderealTime(d,lw){return rad*(280.16+360.9856235*d)-lw}function astroRefraction(h){return h<0&&(h=0),2967e-7/Math.tan(h+.00312536/(h+.08901179))}function solarMeanAnomaly(d){return rad*(357.5291+.98560028*d)}function eclipticLongitude(M){var C=rad*(1.9148*sin(M)+.02*sin(2*M)+3e-4*sin(3*M)),P=rad*102.9372;return M+C+P+PI}function sunCoords(d){var M=solarMeanAnomaly(d),L=eclipticLongitude(M);return{dec:declination(L,0),ra:rightAscension(L,0)}}function julianCycle(d,lw){return Math.round(d-J0-lw/(2*PI))}function approxTransit(Ht,lw,n){return J0+(Ht+lw)/(2*PI)+n}function solarTransitJ(ds,M,L){return J2000+ds+.0053*sin(M)-.0069*sin(2*L)}function hourAngle(h,phi,d){return acos((sin(h)-sin(phi)*sin(d))/(cos(phi)*cos(d)))}function observerAngle(height){return-2.076*Math.sqrt(height)/60}function getSetJ(h,lw,phi,dec,n,M,L){var w=hourAngle(h,phi,dec),a=approxTransit(w,lw,n);return solarTransitJ(a,M,L)}function moonCoords(d){var L=rad*(218.316+13.176396*d),M=rad*(134.963+13.064993*d),F=rad*(93.272+13.22935*d),l=L+rad*6.289*sin(M),b=rad*5.128*sin(F),dt=385001-20905*cos(M);return{ra:rightAscension(l,b),dec:declination(l,b),dist:dt}}function hoursLater(date,h){return new Date(date.valueOf()+h*dayMs/24)}PI=Math.PI,sin=Math.sin,cos=Math.cos,tan=Math.tan,asin=Math.asin,atan=Math.atan2,acos=Math.acos,rad=PI/180,dayMs=864e5,J1970=2440588,J2000=2451545,e=rad*23.4397,SunCalc={},SunCalc.getPosition=function(date,lat,lng){var lw=rad*-lng,phi=rad*lat,d=toDays(date),c=sunCoords(d),H=siderealTime(d,lw)-c.ra;return{azimuth:azimuth(H,phi,c.dec),altitude:altitude(H,phi,c.dec)}},times=SunCalc.times=[[-.833,'sunrise','sunset'],[-.3,'sunriseEnd','sunsetStart'],[-6,'dawn','dusk'],[-12,'nauticalDawn','nauticalDusk'],[-18,'nightEnd','night'],[6,'goldenHourEnd','goldenHour']],SunCalc.addTime=function(angle,riseName,setName){times.push([angle,riseName,setName])},J0=9e-4,SunCalc.getTimes=function(date,lat,lng,height){height=height||0;var lw=rad*-lng,phi=rad*lat,dh=observerAngle(height),d=toDays(date),n=julianCycle(d,lw),ds=approxTransit(0,lw,n),M=solarMeanAnomaly(ds),L=eclipticLongitude(M),dec=declination(L,0),Jnoon=solarTransitJ(ds,M,L),i,len,time,h0,Jset,Jrise,result={solarNoon:fromJulian(Jnoon),nadir:fromJulian(Jnoon-.5)};for(i=0,len=times.length;i<len;i+=1)time=times[i],h0=(time[0]+dh)*rad,Jset=getSetJ(h0,lw,phi,dec,n,M,L),Jrise=Jnoon-(Jset-Jnoon),result[time[1]]=fromJulian(Jrise),result[time[2]]=fromJulian(Jset);return result},SunCalc.getMoonPosition=function(date,lat,lng){var lw=rad*-lng,phi=rad*lat,d=toDays(date),c=moonCoords(d),H=siderealTime(d,lw)-c.ra,h=altitude(H,phi,c.dec),pa=atan(sin(H),tan(phi)*cos(c.dec)-sin(c.dec)*cos(H));return h+=astroRefraction(h),{azimuth:azimuth(H,phi,c.dec),altitude:h,distance:c.dist,parallacticAngle:pa}},SunCalc.getMoonIllumination=function(date){var d=toDays(date||new Date),s=sunCoords(d),m=moonCoords(d),sdist=149598e3,phi=acos(sin(s.dec)*sin(m.dec)+cos(s.dec)*cos(m.dec)*cos(s.ra-m.ra)),inc=atan(sdist*sin(phi),m.dist-sdist*cos(phi)),angle=atan(cos(s.dec)*sin(s.ra-m.ra),sin(s.dec)*cos(m.dec)-cos(s.dec)*sin(m.dec)*cos(s.ra-m.ra));return{fraction:(1+cos(inc))/2,phase:.5+.5*inc*(angle<0?-1:1)/Math.PI,angle:angle}},SunCalc.getMoonTimes=function(date,lat,lng,inUTC){var t=(typeof date)[0]=='o'?date:new Date(date);inUTC?t.setUTCHours(0,0,0,0):t.setHours(0,0,0,0);var hc=.133*rad,h0=SunCalc.getMoonPosition(t,lat,lng).altitude-hc,h1,h2,rise,set,a,b,xe,ye,d,roots,x1,x2,dx;for(var i=1;i<=24;i+=2){if(h1=SunCalc.getMoonPosition(hoursLater(t,i),lat,lng).altitude-hc,h2=SunCalc.getMoonPosition(hoursLater(t,i+1),lat,lng).altitude-hc,a=(h0+h2)/2-h1,b=(h2-h0)/2,xe=-b/(2*a),ye=(a*xe+b)*xe+h1,d=b*b-4*a*h1,roots=0,d>=0&&(dx=Math.sqrt(d)/(Math.abs(a)*2),x1=xe-dx,x2=xe+dx,Math.abs(x1)<=1&&roots++,Math.abs(x2)<=1&&roots++,x1<-1&&(x1=x2)),roots===1?h0<0?rise=i+x1:set=i+x1:roots===2&&(rise=i+(ye<0?x2:x1),set=i+(ye<0?x1:x2)),rise&&set)break;h0=h2}var result={};return rise&&(result.rise=hoursLater(t,rise)),set&&(result.set=hoursLater(t,set)),rise||set||(result[ye>0?'alwaysUp':'alwaysDown']=!0),result},(typeof exports)[0]=='o'&&(typeof module)[0]!='u'?module.exports=SunCalc:(typeof define)[0]=='f'&&define.amd?define(SunCalc):window.SunCalc=SunCalc}()\");\n\u0010\u001b[1dvar i;\n\u0010\u001b[3dfor (i = 0; i < 300; i++) {\u001b\n  print(i);\u001b\n  print(require(\"suncalc\").getTimes(new Date(), 0, 0, 0).sunrise);\u001b\n}\n\n"
Splitting for reset(), delay 250
XOFF received => pause upload
XON received => resume upload
Splitting for Modules.addCached, delay 250
>>> Sent
ERROR: [notify_error] Prompt not detected - upload failed. Trying to recover...
Splitting for Ctrl-C, delay 250
Splitting for Ctrl-C, delay 250
```

nxdefiant, 2023-10-28:
> Prompt not detected - upload failed. Trying to recover...

This looks more like the IDE trying to kill the program. Looks like its checking the prompt after upload which your heavy computation prevents. How about waiting before executing the program?

> setTimeout(() => {
  for (let i = 0; i < 300; i++) {
    print(i);
    print(require("suncalc").getTimes(new Date(), 0, 0, 0).sunrise);
  }
}, 5000);

g-rden, 2023-10-28:
I didn't do enough testing. It seems like an error occurs after some amount of computing?
This also errors:
```
var i;
var x;

for (i = 0; i < 30000; i++) {
  x = new Date();
}

print(i);
print(x);
```

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v19 (c) 2021 G.Williams
>
Prompt not detected - upload failed. Trying to recover...
Execution Interrupted
at line 2 col 16
  x = new Date();
               ^
3664
Date: Sat Oct 28 2023 23:31:13 GMT+0200
>echo(1)
=undefined
```

nxdefiant, 2023-10-28:
Might be related: https://github.com/espruino/EspruinoWebIDE/issues/194

g-rden, 2023-10-28:
> This runs without error on my Bangle.

That does not surprise me :)

> Might be related: https://github.com/espruino/EspruinoWebIDE/issues/194

Hmm maybe.

I changed the program I am working on, so it won't call the function that often. But I am not sure what to do with this issue now.

g-rden, 2023-10-28:
@nxdefiant Waiting beforehand works. Thanks

closed 2023-10-28 by g-rden

gfwilliams, 2023-10-30:
This is expected behaviour - specifically:

> Prompt not detected - upload failed. Trying to recover...

The IDE uploaded some code (likely to RAM?) which gets executed as it's uploaded, but then the IDE detects that it's taking a long time to complete, so rather that leave you with a blank, unresonsive left-hand side of the IDE it tries to break out of the executing code.

If possible, ideally you'd avoid anything that takes that long (even after a delay) as while it won't break out on the watch it can make it seem unresponsive.