espruino/BangleApps #3073: [widclk] String artifact when changing number of digits showing

dormio, 2023-11-03:
### Affected hardware version

Bangle 2

### Your firmware version

0.06

### The bug

The widget was displaying "12:59." When the time changed to 1:00, the display was "1:009," I assume due to the difference in number of characters (therefore not overwriting the 9).

closed 2023-11-03 by gfwilliams

bobrippling, 2023-11-03:
@dormio does #3074 fix this?

(to try it out, you can paste the [new widclk](https://github.com/espruino/BangleApps/blob/9142cf6805bc3df590b58b332307bb41d5acdf6f/apps/widclk/widget.js) into [the IDE](https://www.espruino.com/ide/#) and upload it as `widclk.js`)

gfwilliams, 2023-11-03:
Thanks for the report!

We were both working on this! - I've just pushed a fix to the development app loader.

I think probably when it was originally developed a 24 hour clock was used, so everything stayed the same width - it was only on 12 hour clock where this showed issues

commit 319d22fa (2023-11-03): widclk 0.08: Ensure we clear the whole rect so we don't end up leaving old text when time changes (fix #3073)