espruino/BangleApps #3081: [Recorder] Appending with different sensors causes data in wrong columns

ahormann, 2023-11-05:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

When recording anything and deciding after a while that you don't want to log a sensor anymore (eg to save battery) the resulting csv (downloaded from the app-store) shifts data to wrong columns (eg the charging column contains the altitude).

### Steps to reproduce:
* Start recording with HR, battery, steps and barometer
* Wait for a few datapoints
* Stop recording
* Deselect Battery Log
* Start recording and append to old file
* Wait for a few datapoints
* Stop recording and look into csv to see the bug

Note: This also has the consequence, that when logging GPS before and after appending and exporting the gpx and ignoring datapoints without GPS, many entries without GPS are generated (because GPS is stored as another value).

### Installed apps

openstmap (0.25)
widid (0.03)
antonclk (0.11)
wid_a_battery_widget (0.04)
bthrm (0.17)
widlock (0.08)
agpsdata (0.06)
widmessages (0.05)
messageicons (0.05)
widbt (0.09)
lightswitch (0.09)
android (0.32)
iconlaunch (0.19)
messagegui (0.74)
notify (0.13)
messages (0.62)
boot (0.59)
antonclkplus (0.10)
locale (0.17)
gbridge (0.27)
hrm (0.11)
health (0.27)
recorder (0.32)
setting (0.65)

nxdefiant, 2023-11-06:
@gfwilliams What this done intentionally to save space?

gfwilliams, 2023-11-06:
Not such, it just makes sense to put in columns for only what you're recording - but assume we just put in every column all the time anyway... What happens if you stop recording, then install a new app that provides new data that can be recorded, then resume? same thing.

I guess when we start recording we need to load the first line (headings) of the file, and compare it with our current set of headings. If it's different we don't even provide the option to append?

I'll look into it.

bobrippling, 2023-11-15:
Reopening after [a small bug found](https://github.com/espruino/BangleApps/commit/dd96e33774bee9c0ce0aea9ffe336ce18a0d631c#r132592068)

commit dd96e337 (2023-11-06): 0.33: Ensure that a new file is always created if the stuff that's being recorded has changed (fix #3081)
PR #3098 (2023-11-15): recorder: don't prompt the user when the columns change - force new file
Resolves a small issue with #3081, see [commit comments](https://github.com/espruino/BangleApps/commit/dd96e33774bee9c0ce0aea9ffe336ce18a0d631c#r132592068) for more detail

Closes #3081
closed 2023-11-15 by thyttan