espruino/BangleApps #3086: [app/widget name] Short description of bug

joshaspinall, 2023-11-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Widalt app stops updating the altitude.  Reboot causes the altitude to blank, does not recover.  Did not note when the updates stopped.  Issue persists through power off events.

Widalt settings inputs are ignored.  QNH is 1013, altitude shows 44330m, update interval 5s.

### Installed apps

- Defaults
- widalt

bobrippling, 2023-11-08:
> Issue persists through power off events

This makes it sound like maybe a firmware / hardware issue, @gfwilliams would this seem accurate?

nxdefiant, 2023-11-08:
Try [baroreset.settings.js](https://espruino.microcosm.app/api/v1/files/be4464b77c7470a8b5970b65d08d74e9c371afbf.js) or [baroreset.boot.js](https://espruino.microcosm.app/api/v1/files/a3ed4d09035158425c8705d692ba09f396b656a9.js) attached in the [forum thread](https://forum.espruino.com/conversations/373620/)

gfwilliams, 2023-11-09:
Yes, I think it's a hardware/firmware issue - somehow, rarely, the barometer can get misconfigured. If you flatten the battery (there's an option in the Settings app) then almost certainly the barometer will start working again.

The code in above should work but it's a bit heavy-handed, involving a Bangle.js reset.

The problem is it's been pretty hard to track down as I've never seen this happen on a device here - and when I tried previously resetting the pressure sensor at power on each time didn't work.

I've just made some changes to the main firmware though so if you install the latest cutting edge Espruino firmware it should now work for you.

Now, it automatically resets the sensor. It seems that right after you ask the sensor to reset it reports that it is actually ready (but it's not). You actually need to wait a few milliseconds and *then* it says it's not ready, and then you wait a bit more and eventually it's fine.

joshaspinall, 2023-11-09:
> Do you get a value in the [barometer](https://banglejs.com/apps/?id=barometer)?
> 
> If not try [baroreset.settings.js](https://espruino.microcosm.app/api/v1/files/be4464b77c7470a8b5970b65d08d74e9c371afbf.js) or [baroreset.boot.js](https://espruino.microcosm.app/api/v1/files/a3ed4d09035158425c8705d692ba09f396b656a9.js) attached in the [forum thread](https://forum.espruino.com/conversations/373620/)

Just installed the Barometer, and this is also blank.

Thanks for the responses.  I may install the cutting edge but will more likely wait until the next released version.  Happy to report back but it sounds like you are confident it is resolved. :)

joshaspinall, 2023-11-15:
@gfwilliams I have just updated to the latest Edge build, https://www.espruino.com/binaries/travis/master/espruino_2v19.60_banglejs2.zip, and the issue is still present, as previously described.  I have flattened the battery and the issue persists.  Happy to help, provide info wherever I can.

gfwilliams, 2023-11-16:
Ok, thanks for testing - this is a really strange one. What timezone are you in?

Maybe we could set up remote access for the Web IDE and then I can connect to your watch and have a poke around to see if I can figure out what's going wrong?

joshaspinall, 2023-11-21:
@gfwilliams Indeed!  Hopefully it doesn't turn out to be a red herring or an issue with my specific unit.

I am GMT/UTC based in the UK.  Happy to support if I can.  Please contact me directly via email and we can arrange to suit.

bobrippling, 2023-11-22:
Just reopening this while it's still unsolved @joshaspinall @gfwilliams

bobrippling, 2024-01-13:
Sounds like [the new firmware fixes this](https://forum.espruino.com/comments/17251740/)

closed 2024-01-15 by gfwilliams