espruino/BangleApps #3088: [qmsched] Quiet mode set to "Alarms" does not get turned off when alarm fires

gammore, 2023-11-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Hi!

I have installed the quiet mode app (https://github.com/espruino/BangleApps/tree/master/apps/qmsched) and set up the two timers (8 to turn off and 23 to turn on "alarms" mode). 

My understanding of that feature is the following:
 - If the time to fire a schedule comes, the schedule fires and turns on/off quiet mode.
 - If quiet mode is set to "alarms", the next alarm that fires disables quiet mode.

The schedule part works as expected, when the time comes (8am or 11pm) the watch changes the quiet mode setting, but the alarms mode doesn't.

I've waited for an alarm to fire having the icon on the top (the quiet mode symbol with the little bell) and, when the alarm starts I press stop but the icon doesn't disappear.

I've tried resetting the watch but after installing the minimal apps (alarms and quiet mode) the problem persisted.

If you know anything I could do to help debug this please let me know!

Regards

### Installed apps

- alarms & timers
- bootloader
- launcher
- anton clock
- about
- lock widget
- bluetooth widget
- bluetoot id widget
- welcome
- notifications
- health tracking
- scheduler
- alarms widget
- quiet mode scheduler and widget
- battery level widget
- settings
- message icons
- message ui
- message widget
- messages
- android integration
- pebble clock

bobrippling, 2023-11-11:
Thanks for the report - to check I've got the right end of the stick, is the issue you're seeing that you expect quiet mode to be disabled if an alarm sounds during quiet mode? (if the mode is set to "Alarms")

If so, I don't believe this is intended behaviour of the app currently. Going off the code, the "Alarms" vs. "Silent" setting only affects the widget drawing, specifically this:

https://github.com/espruino/BangleApps/blob/4222ea6e138dd7993895ab8762a1c017293ec180/apps/qmsched/widget.js#L26-L33

Or to summarise, in "Alarms" mode, we get a little alarm icon as well as the no-entry icon

@rigrig does this match your intention for the app?

rigrig, 2023-11-12:
@gammore I'm afraid that's not how it works, "Alarms" simply means that alarms still go off, but everything else is silenced. "Quiet Mode" is [documented](https://github.com/espruino/BangleApps/tree/master/apps/setting#quiet-mode) in the `setting` app, because even without `qmsched` you can always set it manually.

@bobrippling Yeah, this app doesn't do much besides changing settings and drawing the widget. The actual "adhering to quiet mode" is up to individual apps (see e.g. https://github.com/espruino/BangleApps/pull/697 where it was first added)

gammore, 2023-11-12:
Hi!

It seems I assumed the behaviour for the name, I'm sorry!

If that's not how it works, this is clearly not a bug.

Thanks!

closed 2023-11-12 by gammore

bobrippling, 2023-11-13:
Cool, thanks for the clarification / understanding!