espruino/BangleApps #3114: [owmweather] refreshing weather data fails with "Not OWM data"

aglidden, 2023-11-25:
### Affected hardware version

Bangle 2

### Your firmware version

2v19.75

### The bug

Hello. My troubleshooting of [owmweather](https://github.com/espruino/BangleApps/blob/master/apps/owmweather/) led me to #3110. 

The app returns "[Not OWM data](https://github.com/espruino/BangleApps/blob/62686795a559bebfc702e84ebbe0babe5bc54faf/apps/owmweather/lib.js#L31)" when refreshing the weather data. Is this related to it calling `require("Storage").readJSON(`, `require("Storage").writeJSON(`, and `JSON.parse(` in [lib.js](https://github.com/espruino/BangleApps/blob/master/apps/owmweather/lib.js) and the new relaxed JSON parser? 
I installed off of the loader https://banglejs.com/apps/ and I tried the dev version on https://espruino.github.io/BangleApps/ as well (I'm not 100% sure I did that correctly, but it seemed to install correctly from there).

It works fine on firmware 2v19, returning "Success". One should be able to reproduce this by installing Weather and OWM Weather on firmware 2v19.75, installing an openweathermap API key via the app manager, then going to Settings -> Apps -> OWM Weather -> Force refresh

nxdefiant, 2023-11-25:
#3110 is not yet on the official app loader and only affects the interface.html, not the app itself. So I guess its not this change.

aglidden, 2023-11-25:
I was thinking, perhaps erroneously, that it might be a change in the 2v19.75 firmware that #3110 was addressing and might need to be applied to lib.js as well? I don't know. I thought it looked json parsing related though, for what it's worth.

bobrippling, 2023-11-26:
Uncertain for the moment - waiting for my owmweather key to activate. Do you have the JSON you get back from the request to take a look at in the meanwhile?

aglidden, 2023-11-27:
`{"coord":{"lon":-105.2,"lat":39.89},"weather":[{"id":801,"main":"Clouds","description":"few clouds","icon":"02n"}],"base":"stations","main":{"temp":269.05,"feels_like":265.83,"temp_min":265.29,"temp_max":271.62,"pressure":1021,"humidity":68},"visibility":10000,"wind":{"speed":2.06,"deg":330},"clouds":{"all":20},"dt":1701045759,"sys":{"type":2,"id":2004578,"country":"US","sunrise":1701007078,"sunset":1701041913},"timezone":-25200,"id":7150586,"name":"Countryside","cod":200}`

bobrippling, 2023-11-27:
Yeah it's an issue with `JSON.parse` in 2v19.75:
```javascript
>s = '{"coord":{"lon":-105.2,"lat":39.89},"weather":[{"id":801,"main":"Clouds","description":"few clouds","icon":"02n"}],"base":"stations","main":{"temp":269.05,"feels_like":265.83,"temp_min":265.29,"temp_max":271.62,"pressure":1021,"humidity":68},"visibility":10000,"wind":{"speed":2.06,"deg":330},"clouds":{"all":20},"dt":1701045759,"sys":{"type":2,"id":2004578,"country":"US","sunrise":1701007078,"sunset":1701041913},"timezone":-25200,"id":7150586,"name":"Countryside","cod":200}'
=...
>JSON.parse(s)
={ }
```

Strings and arrays are ok, looks like an issue with objects:
```javascript
>JSON.parse('{"c":1}')
={  }
>JSON.parse('["4",5,"six"]')
=[
  "4",
  5,
  "six"
 ]
>JSON.parse('["4",5,"six",{"x":5}]')
Uncaught SyntaxError: Got ':' expected ','
 at line 1 col 18
["4",5,"six",{"x":5}]
```

I believe it's from the early return in [this espruino commit](https://github.com/espruino/Espruino/commit/59dda13c259888671276a55dbcdb7579223d6329) and I've raised fix.

closed 2023-11-27 by gfwilliams

aglidden, 2023-11-27:
Hm, now with 2v19.78 it's saying `[JUncaught undefined` in the debug logs and displaying `SyntaxError: Got '{' expected ':`

[image]

gfwilliams, 2023-11-27:
Ooops, sorry - should have tested that. I just pushed a change that I think should fix it now