espruino/BangleApps #3116: [run v0.17] Distance is wrong

pavelmachek, 2023-11-29:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Distance in kilometers is pretty much wrong.

a) On startup, very frequently large distance is added when acquiring GPS fix. It may be distance to previous GPS fix or something like that.

b) When stationary, random position noise from GPS means that distance is growing quite rapidly. [Possible solution might be to test distance from last point; if it is less then 10m, ignore GPS measurement. if it is more, save current position as last point and add the distance].

c) When user returns to the clock, track recording seems to continue but distance is not accounted properly. Straight-line distance is taken from last point when Run was running, or something like that. [Possible solution might be for "Recorder" to do the distance calculation, and make "Run" ask "Recorder" for distance.]

bobrippling, 2023-11-29:
I like the sound of your proposed fixes, specifically:

- We wait until we have a stable fix at the start
  - Perhaps we wait until two consecutive GPS fixes are less than 10m apart, or more than 30 seconds has passed with more than 3 fixes during this time?
- Like you say - we pause the run/recording if the user hasn't moved over 10m in the past 30 seconds or so, it'd mirror what apps like Strava do for cycling ("auto-pause") and so on
- Your last point sounds like a combination of the two previous - we'd need to detect resuming after a "pause" (2nd point) and then re-acquire regular fixes (1st point)

Does that all sound good to you? If so, we have a plan

pavelmachek, 2023-11-30:
a) Yes, that should do the trick. Waiting for altitude to stop changing may be easiest (we have barometer, so we can tell the altitude is not changing).
b) Yes, stopping recording at low speed would work, too.
c) I'm proposing something else. Do the distance computation in Recorder (not in Run), so that distance is still updated when Clocks are running.

bobrippling, 2023-11-30:
Great, sounds like a plan.

About **c)** (in fact, this applies to all points, but specifically about where the change should be made), I don't believe `Run` records distance, nor does `Recorder` (funnily enough), it's done by the `exstats` module, [setup here in `Run`](https://github.com/espruino/BangleApps/blob/e5ebb03e1385cb0842bf9dcf4579cad45a32b029/apps/runplus/app.js#L52-L52=) and `Recorder` doesn't calculate distance at all, it just notes down each GPS point as it comes in.

So I believe the fix would be to change exstat's [distance calculation](https://github.com/espruino/BangleApps/blob/e5ebb03e1385cb0842bf9dcf4579cad45a32b029/modules/exstats.js#L148-L149), which would then be picked up by `Run`/`Run`plus automatically, since they display the stats.

For exporting runs, we'd need to apply similar logic to the `Recorder` - I think for this we'd want `Recorder` to continue to write raw coordinates, and we apply the logic in [`interface.html`](https://github.com/espruino/BangleApps/blob/e5ebb03e1385cb0842bf9dcf4579cad45a32b029/apps/recorder/interface.html#L280-L288) when we're exporting routes.

bobrippling, 2023-12-01:
gipy's [latest update](https://github.com/espruino/BangleApps/pull/3117) has [some logic](https://github.com/espruino/BangleApps/blob/f4d43072297a7312d1a728c0177b3d983dec0e81/apps/gipy/app.js#L475-L485) around this too

gfwilliams, 2024-10-28:
Just had a play - this appears to do something:

```
Bangle.setGPSPower(1);

const EARTH_RADIUS = 6371008.8;
const POS_ACCURACY = 2.5;
const VEL_ACCURACY = 0.05;

var state = {
  t : Date.now(),
  dt : 0,
  distance : 0
};

function updateGps(gps) {
  const t = Date.now();
  const dt = (t - state.t) / 1000;
  const d2s = Math.PI/180;
  state.t = t;
  state.dt += dt;
  
  if (!gps.fix) {
    return;
  }
  const r = EARTH_RADIUS + gps.alt;
  const x = r * Math.cos(gps.lat*d2s) * Math.cos(gps.lon*d2s);
  const y = r * Math.cos(gps.lat*d2s) * Math.sin(gps.lon*d2s);
  const z = r * Math.sin(gps.lat*d2s);
  const v = gps.speed;
  if (!state.x) {
    state.x = x;
    state.y = y;
    state.z = z;
    state.v = v;
    state.pError = (0|gps.hdop) * POS_ACCURACY;
    state.vError = (0|gps.hdop) * VEL_ACCURACY;
    return;
  }
  const dx = x - state.x;
  const dy = y - state.y;
  const dz = z - state.z;
  const dv = v - state.v;
  const dpMag = Math.sqrt(dx * dx + dy * dy + dz * dz);
  const dvMag = Math.abs(dv);
  state.pError += state.v * state.dt;
  state.dt = 0;
  const pError = dpMag + gps.hdop * POS_ACCURACY;
  const vError = dvMag + gps.hdop * VEL_ACCURACY;
  const pGain = state.pError / (state.pError + pError);
  const vGain = state.vError / (state.vError + vError);
  
  state.x += dx * pGain;
  state.y += dy * pGain;
  state.z += dz * pGain;
  state.v += dv * vGain;
  state.pError += (pError - state.pError) * pGain;
  state.vError += (vError - state.vError) * vGain;
  const pMag = Math.sqrt(state.x * state.x + state.y * state.y + state.z * state.z);
  state.lat = Math.asin(state.z / pMag) / d2s;
  state.lon = (Math.atan2(state.y, state.x) / d2s) || 0;
  state.alt = pMag - EARTH_RADIUS;
  if (true) {
    state.distance += dpMag * pGain;
    state.speed = (state.distance / state.duration) || 0;
    state.cadence = (60 * state.steps / state.duration) || 0;
  }
  print(state);
}
Bangle.on("GPS", updateGps);
```

I'm not sure if it works or not - it might be interesting to take it out somewhere and look at the distance value that's reported vs that of exstats