espruino/BangleApps #3128: [lcars] Temperature is just a black bar

Zombiefleischer, 2023-12-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

I only get a black bar that spans all the way to the left side instead of a temperature reading from the watch face. Other weather related infos are working correctly. It was working for a short while after I reseted the watch to factory settings and installed only the needed apps, but an hour or so later it stopped working so I installed a bunch of other apps. Also to note, it works fine on something like the Weather Clock or the Weather App.
[image]

I use Tiny Weather Forecast as my source and the banglejs gadgetbridge as bridge.

### Installed apps

Anton Clock
About
Lock Widget
Bluetooth ID Widget
Welcome
Scheduler
Alarms Widget
Bootloader
Bluetooth Widget
Notifications (default)
Health Tracking
Settings
Message Icons
Message Widget
Messages
Android Integration
Message List
A Battery Widget (with percentage)
App Manager
Weather
Desktop Launcher
GPS Info
Barometer
Gadgetbridge Music Controls
OpenStreetMap
Thermometer
GPS Time
Languages
Dice-n-Roll
Weather Clock
LCARS Clock
Alarms & Timers
Clockinfo Clock
Clock Info Module
Clock Info Widget

bobrippling, 2023-12-14:
Looks like it might be a colour problem - @peerdavid anything like this ring a bell?

CountMurphy, 2024-01-08:
Its an issue with the 2.19 firmware. I have the same problem. Downgrading to 2.18 fixes it.

bobrippling, 2024-01-08:
Thanks for checking - sounds like it's related to [this espruino commit](https://github.com/espruino/Espruino/commit/eabd13cfb8ce1c0a81708bb6e9f37f15b101a5fd#commitcomment-135446714) and #3144, so should be fixed when [this commit](https://github.com/espruino/Espruino/commit/91e45d60f1693ed61c93c0163b7c30d6168cc7d9) makes it to firmware (v2.20 at a guess)

CountMurphy, 2024-01-08:
I'll report back whenever 2.20 comes out

CountMurphy, 2024-02-19:
2.20 still has the issue.

bobrippling, 2024-02-19:
I guess the [above commit](https://github.com/espruino/BangleApps/issues/3128#issuecomment-1881847696) wasn't the fix, looks like the bug would need some more scrutiny

peerdavid, 2024-02-20:
Hi there -- I jsut updated and its working on my Bangle so I can't reproduce right now. But to search the bug, I honestly guess its the weather app and not LCARS as LCARS uses the json from the weather app. Especially as of the report that " FWIW I'm using the "weather" app and its broken there too."...

CountMurphy, 2024-02-20:
Hmm, wonder if I have some kind of corrupted setting/config file. I'll update again later this week and do a full factory reset to see if it still shows up.

lawern, 2024-10-15:
See if it works for you now! I think I corrected that in June: #3461

closed 2024-10-17 by thyttan