espruino/BangleApps #3135: [Recorder] Day in filename is often wrong

Mineinjava, 2023-12-20:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

1. Open recorder
2. Start recording
3. Stop recording after a few data points have been recorded
4. view the file that was created

From my experience, about 50% of the recording files have a name that is a day later than the date it was recorded on.

My timezone on watch is set to GMT-8. This is likely the cause of the issue. 

[image]

### Installed apps

widlock (0.08), sched (0.22), sched (0.22), widalarm (0.01), messageicons (0.05), boot (0.59), notify (0.13), clock_info (0.07), gpsinfo (0.12), bikespeedo (0.06), gpsnav (0.08), dtlaunch (0.24), bwclk (0.33), widmsggrid (0.05), widminbt (0.01), widgps (0.09), slides (undefined), magnav (0.06), quicklaunch (0.15), runplus (0.22), kbswipe (0.09), clkinfostopw (0.03), mylocation (0.10), clkinfosunrise (0.03), configurable_clock (0.03), locale (0.17), sleeplog (0.15), widclk (0.08), gpsmagcourse (0.01), waypoint_editor (0.03), rebbleagenda (0.01), weather (0.26), clkinfocal (0.02), ios (0.15), waypoints (0.03), setting (0.68), messagegui (0.75), messages (0.62), fastload (0.06), fastreset (0.03), flashcount (0.01), widbatv (0.03), alarm (0.46), health (0.28), openstmap (0.27), recorder (0.37), myprofile (0.01)

nxdefiant, 2023-12-20:
I think storing the date as UTC is a good choice, but I would try to display the date in local time, or?

Btw: Do you really have installed sched two times?!?

Mineinjava, 2023-12-20:
The issue I was having was with the file names. The recorder wouldn't append to the file because it expected the file to have a different name and would instead create a new file.

Say, for example, I start recording at 23:35 utc (15:35 local time). I then stop recording at 23:55 (15:55). I record again at 00:05 (16:05), and the recorder creates a new file instead of appending.

Yes, It would appear I have sched installed twice. I'm not sure how this happened and I'm a bit afraid to remove one of them.

bobrippling, 2023-12-20:
For recorder, I propose we fix this with a change to the logic - currently we'll see if there's [any logs for the current day](https://github.com/espruino/BangleApps/blob/ba082f670c5661894a83c4bb666fcea555d85aec/apps/recorder/widget.js#L260-L264). Instead I propose that we look at the most recent log and if it's within 4 hours (for example), we offer to resume that log.

This would also fix the issue for anyone (regardless of their timezone) who starts recording before midnight and wants to resume said recording after midnight. What do you both think?

------------

Unrelated - @Mineinjava, for `sched` being installed twice, would you mind running this in [the IDE](https://www.espruino.com/ide/#) and pasting the output here?

```javascript
S=require("Storage");
S.list(/\.info$/).forEach(fnam => {
  var j = S.readJSON(fnam,1)||{};
  console.log(JSON.stringify({id:fnam.slice(0,-5),ver:j.version,fnam}));
})
```

Mineinjava, 2023-12-20:
Four hours seems reasonable. However it may be useful to make it configurable.

Mineinjava, 2023-12-20:
> Unrelated - @Mineinjava, for `sched` being installed twice, would you mind running this in [the IDE](https://www.espruino.com/ide/#) and pasting the output here?

This outputs:

<details>

<summary> Long code block </summary>

```
{"id":"widlock","ver":"0.08","fnam":"widlock.info"}
{"id":"sched","ver":"0.22","fnam":"sched.info"}
{"id":"sched","ver":"0.22","fnam":"sched.info"}
{"id":"widalarm","ver":"0.01","fnam":"widalarm.info"}
{"id":"messageicons","ver":"0.05","fnam":"messageicons.info"}
{"id":"boot","ver":"0.59","fnam":"boot.info"}
{"id":"notify","ver":"0.13","fnam":"notify.info"}
{"id":"clock_info","ver":"0.07","fnam":"clock_info.info"}
{"id":"gpsinfo","ver":"0.12","fnam":"gpsinfo.info"}
{"id":"bikespeedo","ver":"0.06","fnam":"bikespeedo.info"}
{"id":"gpsnav","ver":"0.08","fnam":"gpsnav.info"}
{"id":"dtlaunch","ver":"0.24","fnam":"dtlaunch.info"}
{"id":"bwclk","ver":"0.33","fnam":"bwclk.info"}
{"id":"widmsggrid","ver":"0.05","fnam":"widmsggrid.info"}
{"id":"widminbt","ver":"0.01","fnam":"widminbt.info"}
{"id":"widgps","ver":"0.09","fnam":"widgps.info"}
{"id":"slides","fnam":"slides.info"}
{"id":"magnav","ver":"0.06","fnam":"magnav.info"}
{"id":"quicklaunch","ver":"0.15","fnam":"quicklaunch.info"}
{"id":"runplus","ver":"0.22","fnam":"runplus.info"}
{"id":"kbswipe","ver":"0.09","fnam":"kbswipe.info"}
{"id":"clkinfostopw","ver":"0.03","fnam":"clkinfostopw.info"}
{"id":"mylocation","ver":"0.10","fnam":"mylocation.info"}
{"id":"clkinfosunrise","ver":"0.03","fnam":"clkinfosunrise.info"}
{"id":"configurable_clock","ver":"0.03","fnam":"configurable_clock.info"}
{"id":"locale","ver":"0.17","fnam":"locale.info"}
{"id":"sleeplog","ver":"0.15","fnam":"sleeplog.info"}
{"id":"widclk","ver":"0.08","fnam":"widclk.info"}
{"id":"gpsmagcourse","ver":"0.01","fnam":"gpsmagcourse.info"}
{"id":"waypoint_editor","ver":"0.03","fnam":"waypoint_editor.info"}
{"id":"rebbleagenda","ver":"0.01","fnam":"rebbleagenda.info"}
{"id":"weather","ver":"0.26","fnam":"weather.info"}
{"id":"clkinfocal","ver":"0.02","fnam":"clkinfocal.info"}
{"id":"ios","ver":"0.15","fnam":"ios.info"}
{"id":"waypoints","ver":"0.03","fnam":"waypoints.info"}
{"id":"setting","ver":"0.68","fnam":"setting.info"}
{"id":"messagegui","ver":"0.75","fnam":"messagegui.info"}
{"id":"messages","ver":"0.62","fnam":"messages.info"}
{"id":"fastload","ver":"0.06","fnam":"fastload.info"}
{"id":"fastreset","ver":"0.03","fnam":"fastreset.info"}
{"id":"flashcount","ver":"0.01","fnam":"flashcount.info"}
{"id":"widbatv","ver":"0.03","fnam":"widbatv.info"}
{"id":"alarm","ver":"0.46","fnam":"alarm.info"}
{"id":"health","ver":"0.28","fnam":"health.info"}
{"id":"openstmap","ver":"0.27","fnam":"openstmap.info"}
{"id":"recorder","ver":"0.37","fnam":"recorder.info"}
{"id":"myprofile","ver":"0.01","fnam":"myprofile.info"}
```
</details>

bobrippling, 2023-12-20:
I've got a draft fix (#3136), if you'd like to try it out, it's deployed to my [gh pages](https://bobrippling.github.io/BangleApps/?id=recorder).

I think the four hours logic will want a setting, but let's see how well this works.

----

> > Unrelated - @Mineinjava, for `sched` being installed twice, would you mind running this in [the IDE](https://www.espruino.com/ide/#) and pasting the output here?
> 
> This outputs:
> Long code block
> 
> ```
> ...
> {"id":"sched","ver":"0.22","fnam":"sched.info"}
> {"id":"sched","ver":"0.22","fnam":"sched.info"}
> ...
> ```

Thanks for that - a file shouldn't appear twice so I'd guess this is a partially corrupt storage situation. You might want to backup anything important on your watch. @gfwilliams what do you think?

Mineinjava, 2023-12-20:
That file has been that way since I got the watch. Weird.

stweedo, 2023-12-20:
The duplicate scheduler app can be fixed by doing a backup and restore from the app loader. There's more info on it here https://forum.espruino.com/conversations/388915/