espruino/BangleApps #3138: [qmsched] Default LCD brightness set to 0.5 causes LCD light up to default value during quiet mode when switching apps

Ishidres, 2023-12-26:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

### Description and how to reproduce
1. Set Settings -> System -> LCD brightness to 0.5
2. Set Quiet Mode LCD brightness to 0.1 => i.e. during normal day schedule the brightness is 0.5 but during quiet mode it should be 0.1
3. Wait for quiet mode to turn on based on schedule (e.g. 22:00 to 08:00)
4. Switch between apps, e.g. go to the clock app, awake the screen using button, go to the desktop launcher, open a new app, etc.
5. You will notice that whenever you switch an app the LCD brightness shortly switches to the "normal day brightness" (0.5) and after around half a second it switches back to the "quiet mode brightness" (0.1)
6. This behavior does *not* occur if normal brightness is set to 1.0 and quiet mode brightness is set to 0.1 => in this case the brightness stays at 0.1 when switching apps during quiet mode (as expected)

### Expected behavior
The LCD brightness should stay unchanged even when the normal day brightness is != 1.0 (e.g. 0.5).

Might be worth testing this on a brand new installation with only qmsched installed.

### Installed apps

boot (0.59), antonclk (0.11), widlock (0.08), notify (0.13), sched (0.22), widalarm (0.01), widbat (0.12), messageicons (0.05), widmessages (0.05), messages (0.62), android (0.32), hrm (0.11), calculator (0.07), quicklaunch (0.15), mylocation (0.10), clkinfosunrise (0.03), clock_info (0.07), authentiwatch (0.07), launch (0.20), circlesclock (0.25), dtlaunch (0.24), sleeplog (0.15), locale (0.17), agenda (0.14), activityreminder (0.11), sleepphasealarm (0.16), slevel (0.04), touchtimer (0.06), messagesmusic (0.05), weather (0.26), run (0.17), messagegui (0.75), setting (0.68), thermom (0.07), owmweather (0.03), kbmulti (0.07), alarm (0.46), ha (0.10), tetris (0.03), bee (0.03), health (0.28), chargent (0.05), intervalTimer (0.01), qmsched (0.10)

bobrippling, 2023-12-30:
Might be to do with the default being applied:

https://github.com/espruino/BangleApps/blob/75970a1aeeb83580f663d2952272c91a28e4cea9/apps/qmsched/lib.js#L78-L80

or this timeout interfering, if you're quick enough to exit:

https://github.com/espruino/BangleApps/blob/2b8399bb9bd076cae54d675baebf7f89b1eeb8d2/apps/qmsched/app.js#L237-L245

@rigrig @halemmerich tagging in case you have any ideas

rigrig, 2023-12-30:
I'm not sure what exactly is causing this issue, but I suspect it has something to do with the brightness first being changed to 0.5 by the general boot (which doesn't happen if it's set to 1.0), and then changed again by the qmsched boot.
We *could* dig into this problem, but I think it'd be better to look at doing [this](https://github.com/espruino/BangleApps/issues/2246#issuecomment-1329851852): replace qmsched with a general "setting schedule" app, so all the "temporary settings while DnD is active" stuff goes away.

Ishidres, 2023-12-30:
> but I think it'd be better to look at doing this: replace qmsched with a general "setting schedule" app, so all the "temporary settings while DnD is active" stuff goes away.

Agree, it would be very nice and helpful if this was already a built-in option and it would also solve this issue.