espruino/BangleApps #3151: [Health Tracking] View data remote

d3nd3, 2024-01-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

When I click Health Tracking Floppy disk icon within BangleApps via GadgetBridge app on android, it does not load, just says loading, please wait.  But within chrome app, it works.  Bug?

### Installed apps

android integration + default.

bobrippling, 2024-01-10:
Can you share your browser's console output when this happens? (F12 to open it)

d3nd3, 2024-01-11:
It's android issue, so no keyboard.

bobrippling, 2024-01-11:
@gfwilliams any ideas? I don't have gadget bridge myself

gfwilliams, 2024-01-11:
If you've got your phone plugged in to your desktop with debugging enabled you can go to `chrome://inspect` on your desktop and can inspect what's happening in your phone's browser. That'd be the way forward.

I just tried it here, and something is just really weird. It's sending the command to get data, and getting all the data back, but somehow it's getting is duplicated.

For example in the console, you can paste:

```
_bangleRx = bangleRx;
bangleRx = function(d) { console.log("DATA",d, connection.received); _bangleRx(d)}
```

So now you get a log to the console every time we get data back (including the accumulated data line), and I see:

```
DATA ["health-2023-6.raw" 
DATA ,"health-2023-7.raw" ["health-2023-6.raw"["health-2023-6.raw"
DATA ,"health-2023-8.raw" ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw"
DATA ,"health-1970-1.raw" ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw"
DATA ,"health-2023-9.raw" ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw"
DATA ,"health-2023-10.raw ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw","health-2023-9.raw","health-2023-9.raw"
DATA ","health-2023-11.ra ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw","health-2023-9.raw","health-2023-9.raw","health-2023-10.raw,"health-2023-10.raw
DATA w","health-2023-12.r ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw","health-2023-9.raw","health-2023-9.raw","health-2023-10.raw,"health-2023-10.raw","health-2023-11.ra","health-2023-11.ra
DATA aw","health-2024-1.r ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw","health-2023-9.raw","health-2023-9.raw","health-2023-10.raw,"health-2023-10.raw","health-2023-11.ra","health-2023-11.raw","health-2023-12.rw","health-2023-12.r
DATA aw"]
 ["health-2023-6.raw"["health-2023-6.raw","health-2023-7.raw","health-2023-7.raw","health-2023-8.raw","health-2023-8.raw","health-1970-1.raw","health-1970-1.raw","health-2023-9.raw","health-2023-9.raw","health-2023-10.raw,"health-2023-10.raw","health-2023-11.ra","health-2023-11.raw","health-2023-12.rw","health-2023-12.raw","health-2024-1.raw","health-2024-1.r
<BLE> Unable to decode "[\"health-2023-6.raw\"[\"health-2023-6.raw\",\"health-2023-7.raw\",\"health-2023-7.raw\",\"health-2023-8.raw\",\"health-2023-8.raw\",\"health-1970-1.raw\",\"health-1970-1.raw\",\"health-2023-9.raw\",\"health-2023-9.raw\",\"health-2023-10.raw,\"health-2023-10.raw\",\"health-2023-11.ra\",\"health-2023-11.raw\",\"health-2023-12.rw\",\"health-2023-12.raw\",\"health-2024-1.raw\",\"health-2024-1.raw\"]\r", got SyntaxError: Expected ',' or ']' after array element in JSON at position 20 (line 1 column 21)
```

Actually nevermind - I fixed it!

closed 2024-01-11 by gfwilliams

d3nd3, 2024-01-11:
I was able to get console using chrome://inspect and android debugging : this is the error: 
`Uncaught TypeError: Cannot read properties of null (reading 'map')
    at interface.html:65:19
    at interface.js:163:13` Hope you fixed it anyway.

gfwilliams, 2024-01-11:
I think if you try now it's live on banglejs.com/apps and should be ok

d3nd3, 2024-01-11:
Yes it is working.

commit 932d2c0d (2024-01-11): On Gadgetbridge, ensure the default data handler is set rather than built into bangleRx.
This means that when comms.js does its hacky thing to replace the data handler it doesn't end up duplicating handlers.

Fixes #3151