espruino/BangleApps #3166: [app/openstmap] Unable to upload maps on 2.20 and master

michaellmonaghan, 2024-01-26:
### Affected hardware version

Bangle 2

### Your firmware version

2v20

### The bug

Only part of the map is displayed when running firmware 2v20. The first several tiles are displayed, then part of a tile is displayed, then none of the following tiles display. Panning and zooming does not cause missing tiles or part of tile to be displayed. Using the same map image file, the full map can be displayed when running firmware 2v19. The same behavior occurs after a factory reset with no other apps installed and when uploading the map image file on firmware 2v19 before upgrading to 2v20. The map size used was XL (10x10) but the same behavior can be observed for other map sizes.
### 2v20
[image]
### 2v19
[image]

### Installed apps

boot (0.59), launch (0.20), antonclk (0.11), about (0.14), widlock (0.08), widbat (0.12), widbt (0.09), notify (0.13), health (0.28), setting (0.68), sched (0.22), widalarm (0.01), alarm (0.46), messageicons (0.05), messagegui (0.75), widmessages (0.05), android (0.32), clock_info (0.07), bwclk (0.33), 93dub (0.06), mtnclock (0.04), weather (0.26), weatherClock (0.06), locale (0.17), waypoints (0.03), gpsnav (0.08), gpsautotime (0.04), speedalt (0.13), recorder (0.37), hrm (0.11), messages (0.62), thermom (0.07), game1024 (0.11), random (0.02), boxclk (0.05), menuwheel (0.02), openstmap (0.27)

nxdefiant, 2024-01-26:
I can reproduce this. For an exaple 5x5 tile I can not access the higher frames, e.g.
```
g.reset().clear();
const img = require("Storage").read("openstmap.2.img");
g.drawImage(img, 0, 0, { "scale": 1.5, "frame": 19 });
```
on 2v19 produces
[image]

and on 2v20 produces
[image]

I think its best to pass this to @gfwilliams

gfwilliams, 2024-01-29:
Ok, thanks - just tried it here and I see the same thing. 

Sorry about this - I'm not sure how that slipped through. It's a strange one as I can't remember anything in particular that would have caused something like this, but I'll look into it in the next few hours

gfwilliams, 2024-01-29:
I've just found the issue - it was some checks in https://github.com/espruino/Espruino/commit/e893f3f56061aafa60d66b259c53bd9d83b4cff2 that restricted the maximum length of flash-memory-mapped strings.

If you update to a cutting-edge build then this should all be fixed now - Unfortunately it looks like I'll need to push out a 2v21 firmware release pretty soon now

closed 2024-01-29 by gfwilliams