espruino/BangleApps #3167: [app/openstmap] Map tiles seams visible depending on location

michaellmonaghan, 2024-01-26:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

When panning, images may overlap by one pixel creating vertical and horizontal seams which are exaggerated by dithering. The seam locations (when present) are identical for each panned location.

A vertical seam is visible in the center of the screen of the first screenshot. The seam disappears when panning slightly to the left.
[image]
[image]

### Installed apps

boot (0.59), launch (0.20), antonclk (0.11), about (0.14), widlock (0.08), widbat (0.12), widbt (0.09), notify (0.13), health (0.28), setting (0.68), sched (0.22), widalarm (0.01), alarm (0.46), messageicons (0.05), messagegui (0.75), widmessages (0.05), android (0.32), clock_info (0.07), bwclk (0.33), 93dub (0.06), mtnclock (0.04), weather (0.26), weatherClock (0.06), locale (0.17), waypoints (0.03), gpsnav (0.08), gpsautotime (0.04), speedalt (0.13), recorder (0.37), hrm (0.11), messages (0.62), thermom (0.07), game1024 (0.11), random (0.02), boxclk (0.05), menuwheel (0.02), openstmap (0.27)

bobrippling, 2024-01-26:
@nxdefiant seems like an off by one, anything sound likely to you?

nxdefiant, 2024-01-26:
Could it be the diffusion for the 3 bit color conversion?

This is what it looks like zoomed in on my device (first + second map):
[image]

This is only the first map drawn:
[image]

This is only the second map drawn:
[image]

On this high zoom level one can clearly see the diffusion pattern. 

@michaellmonaghan Does this also happen for you when unselecting the 3 bit checkbox when uploading maps?

nxdefiant, 2024-01-27:
On another thought about the off-by-one: I believe https://github.com/espruino/BangleApps/blob/master/apps/openstmap/openstmap.js#L79 could use a Math.round(). Maybe other draw statements also.

michaellmonaghan, 2024-01-28:
I am thinking this is a rounding or truncation related error as well.

> @michaellmonaghan Does this also happen for you when unselecting the 3 bit checkbox when uploading maps?

Yes, the tiles will overlap by one pixel either horizontally or vertically, though the effect is much less noticeable.

nxdefiant, 2024-01-28:
ok, could you try the version from my repository?
https://nxdefiant.github.io/BangleApps/index.html?id=openstmap

michaellmonaghan, 2024-01-28:
That did the trick! No overlap with your patch @nxdefiant. Nice job!

PR #3169 (2024-01-30): openstmap: Fix rounding errors
- Fixes #3167 
- interface.html: Fix show maps after delete
- interface.html: sort maps
- interface.html: Load leaflet after all maps are painted which avoids the timeout
- interface.html: Show map bounds according to uploaded size.