espruino/BangleApps #3174: [android] Alarms created from GadgetBridge not repeating

thinkpoop, 2024-01-31:
### Affected hardware version

Bangle 2

### Your firmware version

2v20

### The bug

If you create an alarm in GadgetBridge's UI set to repeat some days (e.g. "MTWTF"), the android integration app doesn't set "rp:true" on the alarm, so the alarm doesn't repeat and is disabled the first time you dismiss it.

The Alarm app on the watch does seem to set rp:true if DOW wasn't left blank. 

A quick fix might be if "dow" from GB is blank, then leave rp:false, but otherwise set rp:true? Somewhere around here in android/boot.js:
https://github.com/espruino/BangleApps/blob/404f34a865201b0d15686f909822e2aae1b7a2cd/apps/android/boot.js#L84C11-L84C74

### Installed apps

boot (0.59), launch (0.20), antonclk (0.11), widlock (0.08), notify (0.13), health (0.28), setting (0.68), sched (0.22), alarm (0.46), locale (0.17), messageicons (0.05), widmessages (0.05), messages (0.62), android (0.32), weather (0.26), barometer (0.04), calendar (0.17), sleeplog (0.15), calculator (0.07), qmsched (0.10), weatherClock (0.06), widalarmeta (0.10), quicklaunch (0.15), qcenter (0.04), owmweather (0.03), chargerot (0.02), agenda (0.14), rebbleagenda (0.01), files (0.08), rgb (0.01), compass (0.09), widminbt (0.01), openstmap (0.27), mylocation (0.10), lcars (0.30), messagegui (0.75), messagesmusic (0.05)

bobrippling, 2024-01-31:
Thanks for the report - sounds like we'd need someone who has GB to try out that quick fix

thyttan, 2024-01-31:
Relates to #2495

thyttan, 2024-01-31:
@thinkpoop would you be up to do a PR with your quick fix proposal to Gadgetbridge's repo?

I could try to give some pointers if you'd like 👍

closed 2024-01-31 by thinkpoop