espruino/BangleApps #3179: Graphics.floodFill behaviour

bobrippling, 2024-02-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

### Description

I'm attempting to use [`g.floodFill`](https://www.espruino.com/ReferenceBANGLEJS2#t_l_Graphics_floodFill) to fill an area inside a circle, marked out by two lines.

### Expected behaviour

The flood fill to fill in the part of the circle marked by the two lines. I see this from a `g.dump()`:

[image]

But on an actual device, pixels in the red circle below ~4/5ths of the height of the `appRect` are not filled. There's also an artifact following the first black line outside of the circle, too, which is visible in the above image.

### Steps to reproduce

```javascript
R = Bangle.appRect;

circ = (pct) => (
  g
  .setColor("#f00")
  .fillCircle(R.x + R.w/2, R.y+R.h/2, R.w * pct)
);

g.clearRect(Bangle.appRect);

circPct = 0.4;
circ(circPct);

g.setColor("#555")
  .drawLine(
    0, R.h * 4/5,
    R.w * circPct - 25, R.h/2
  )
  .drawLine(
    R.w * circPct - 25, R.h/2,
    R.w * 4/5, R.h * 4/5
  )
  .floodFill(
    R.w * circPct - 25,
    R.h/2 + 1
  );
```

### Installed apps

This runs with no installed apps

bobrippling, 2024-02-05:
@gfwilliams is this something you've seen before?

gfwilliams, 2024-02-06:
Ahh, just tried here and I can see the issue - flood fill isn't updating the 'modified area' so the actual image buffer is being set correctly but it's not all being sent out to the display.

You can force it with `g.flip(1)` and the whole image should appear. I'll try and get a fix in, but as I just released version 2v21 it'll be a while before there's a 2v22 with the fix in (although obviously cutting edge should work)

bobrippling, 2024-02-06:
Ah cool, thanks for the quick fix