espruino/BangleApps #3243: [BTHome] Sents multiple events

pinq-, 2024-03-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v21.31

### The bug

Front menu buttons trigger multiple or same event on home assistant. (in the image button 4 and button 5 are trigger)
[image]
reproduce:
If you first press one button in front, it only triggers one event, but if you press second button after that, it triggers both. 
Expected behavior
Every button triggers individual event. 

### Installed apps

launcer, touche keyboard, BTHome, Clock info, pebble++, Rings, Run, Scheduler, welcome.

bobrippling, 2024-03-08:
@gfwilliams I think this is one of yours, maybe intentional behaviour?

gfwilliams, 2024-03-08:
Thanks for the report! How long did you leave it between pressing buttons 4 and 5?

I'm just looking at the code and it seems we do cancel the button press, but only after 30 seconds of inactivity - so if you pressed a second button within that time it'd end up re-reporting both.

I've just fixed that I hope so if you update to 0.02 it should be fixed!

closed 2024-03-08 by gfwilliams