espruino/BangleApps #3251: Error using the Layout library with Typscript

paul-arg, 2024-03-11:
Hello,

I created a new `apps/testApp` folder, with a `testApp.app.ts` file inside.

I am trying to recreate the Layout tutorial code in Typescript so I have

```
const layout: Layout.Layout = new Layout.Layout( {
  type:"v", c: [
    {type:"h", c: [
      {type:"txt", font:"12x20:2", label:"One", pad:4 },
      {type:"txt", font:"12x20:2", label:"Two", pad:4 }
    ]},
    {type:"txt", font:"12x20:2", label:"Three", fillx:1 },
    {type:"h", c: [
      {type:"txt", font:"12x20:2", label:"4", fillx:1 },
      {type:"txt", font:"12x20:2", label:"5", fillx:2 }
    ]},
  ]
});
g.clear();
layout.render();
layout.debug();
```

But VSCode throws an error for `render()` and `debug()`:

`Property 'render' does not exist on type 'new <T extends Hierarchy>(hier: T, options?: { lazy: boolean; btns?: { label: string; cb: () => void; cbl: () => void; }[]; back?: () => void; remove?: () => void; } | undefined) => Layouter<T>'.ts(2339)`

Am I doing something wrong?
Thank you for your help

paul-arg, 2024-03-11:
I found what the problem was.
I needed to define `ids` for each node of the Hierarchy object.

The code becomes:

```
let testLayout = new Layout.Layout(
  {
    type: "v", c: [
      {
        type: "h", c: [
          { type: "txt", font: "12x20:2", label: "One", pad: 4 , id:"a"},
          { type: "txt", font: "12x20:2", label: "Two", pad: 4 , id:"b"}
        ]
      },
      { type: "txt", font: "12x20:2", label: "Three", fillx: 1 , id:"c"},
      {
        type: "h", c: [
          { type: "txt", font: "12x20:2", label: "4", fillx: 1 , id:"d"},
          { type: "txt", font: "12x20:2", label: "5", fillx: 2 , id:"e"}
        ]
      },
    ]
  }
)

testLayout.render();
```

paul-arg, 2024-03-11:
Sorry, now I have a new problem.

I am trying to access `testLayout.a.label` but I have a bunch of errors: 

`'testLayout.a' is possibly 'undefined'.`

`Property 'a' comes from an index signature, so it must be accessed with ['a'].`

```
Property 'label' does not exist on type '{ type: "txt"; font: "Dennis8"; label: string; pad: number; id: string; } | { type: "txt"; font: "12x20:2"; label: string; pad: number; id: string; } | { type: "h"; c: ({ type: "txt"; font: "Dennis8"; label: string; pad: number; id: string; } | { ...; })[]; id: string; font?: never; label?: never; fillx?: never; } |...'.
  Property 'label' does not exist on type '{ type: "v"; c: ({ type: "h"; c: ({ type: "txt"; font: "Dennis8"; label: string; pad: number; id: string; } | { type: "txt"; font: "12x20:2"; label: string; pad: number; id: string; })[]; id: string; font?: never; label?: never; fillx?: never; } | { ...; } | { ...; })[]; id: string; }'.ts(2339)
```

Could someone give me a snippet of code that replicates the JS examples? https://www.espruino.com/Bangle.js+Layout

bobrippling, 2024-03-11:
Glad you solved the first one, for the second issue I've got an MR in progress which will fix it but I'm out of time. It'll be a little while until I can get a solution. In the meanwhile, you can use `testLayout.a!.label` as a workaround though - let me know if you run into anything else.

bobrippling, 2024-03-16:
@paul-arg if you rebase on #3252, does that work? (or alternatively, cherry-pick the changes across)

paul-arg, 2024-03-18:
@bobrippling 
Sorry I won't be able to check this week.
But my test was very simple: declaring `testLayout` and then trying to access `testLayout.a.label`

bobrippling, 2024-03-20:
No worries! Thanks for the test, sounds good. I'll merge it and @paul-arg if you see any issues we can adjust further

PR #3252 (2024-03-20): Fix typescript for `Layout`
`ExtractIds` wasn't picking out the IDs (`id: ... extends string` rather than `id?: ... extends string`) and the resulting union of objects needed to be merged.

Closes #3251

Confirmed with a demo app and `apps/rep` (no version bump as changes are type-level only -> no `.js` changes)
closed 2024-03-20 by bobrippling