espruino/BangleApps #3257: Blood glucose widget Xdrip has stopped displaying

tompusey, 2024-03-13:
Since the 0.03 update the widget no longer displays

atjn, 2024-03-13:
Sorry about that! I caught the mistake and have a fix for it in #3214

If that PR is not merged tomorrow, I can create a separate pull request for it :)

bobrippling, 2024-03-13:
Thanks for the report - @atjn do you know how this got past the linter? I'd expect it to pick up on `p180` not being defined

tompusey, 2024-03-13:
Thanks for looking into this, much appreciated.

gfwilliams, 2024-03-14:
@atjn do you have any idea why the linter would have thrown up and error and requested you to make this change? 

It would be really good to know, as if we found out we might be able to run it again before #3213 was merged and see if it broke any other apps

closed 2024-03-14 by gfwilliams

atjn, 2024-03-14:
@gfwilliams the linter was complaining about the variables `r`, `x` and `y` that are defined underneath `p180`.

It was not a linting error, I made the mistake of also commenting out `p180` despite it not being necessary. I remember how I made the mistake and it required quite a few things to accidentally happen at the same time. I don't think there is any need to go into detail, but one of the unlucky accidents was that I am used to work in an environment where all `eslint:recommended` rules are enabled, and in that case the linter would have always caught this mistake. #3214 enables the rule that would have caught my mistake.

atjn, 2024-03-14:
The good news is that this was the only bug I found (that was made by me) when going through #3214, which suggests to me that there are no other or at least only a few other bugs introduced.

tompusey, 2024-03-14:
That's fixed it, thanks all. Being able to see my blood sugar at a glance is unbelievably helpful.

gfwilliams, 2024-03-14:
@tompusey glad it's fixed! Nice to see the Bangle being used for something like this where it's not just a fun toy but something genuinely helpful :)

@atjn thanks! I kind of assumed you'd got the Linter to auto-fix this, rather than doing it all yourself! That's an immense amount of work, so thanks for all the time you're putting into this!

atjn, 2024-03-14:
@gfwilliams thanks for acknowledging that, it means a lot :) Unfortunately the linter cannot auto-fix any of the rules I have enabled, so it's been a lot of work, but I am happy to help!

commit 5d5c307b (2024-03-14): 0.04: Fixes after regression in 0.03 (fix #3257)