espruino/BangleApps #3260: [BTHome] Sents "ghoste" events and one extra icon

pinq-, 2024-03-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v21.31

### The bug

When button is pressed in front, some times it triggers again without pressing the button so the lights turn back one. I can't find relativity why it dose that or when it dose that. For now, it seems to be completely random for me. 

I have two buttons (light and fan) at front menu, but for some reason, there is two light buttons. The first light button is first in the list and the second light and fan are "side by side". Those two lights also trigger two different buttons. I can't also delete the only one light button. I can delete all the other buttons. 

### Installed apps

launcer, touche keyboard, BTHome, Clock info, pebble++, Rings, Run, Scheduler, welcome.

pinq-, 2024-03-27:
ok, so it is still happening. All the switches do the same. Sometimes multiple times ( I turn on, it turns it off, I turn it on, it turns it off, I trun it on, it stays on). It doesn't happen all the time. Sometimes it works like it should. 
Sometimes I also get this message at the bottom of the screen in red "memory_busy". It goes away when I hard reset it. Don't know if related

bobrippling, 2024-03-27:
Is this similar in behaviour to #3243? Or @gfwilliams would you say it's more of the hardware and light touches? The lights being displayed will be the config file, which it sounds like you've got to the bottom of. If you check [the IDE](https://www.espruino.com/ide/#), are there errors about memory logged?

gfwilliams, 2024-03-28:
> I have two buttons (light and fan) at front menu, but for some reason, there is two light buttons. The first light button is first in the list and the second light and fan are "side by side".

I'm not sure I understand how the icons can be side by side - any chance of a screenshot? App Loader -> More... -> Screenshot

And also connect with the Web IDE and download your `bthome.json` file so I can see how it's set up and reproduce.

It would be helpful if you could attach the log from Home Assistant for the Bangle when this happens - it might explain what is happening.

If you have just a single button, does this happen? Or is it only with multiple buttons?

pinq-, 2024-03-31:
I have the pebble++ front. In there, there is is this "button" that you can slide the options what you see ( steps, bpm, buttons etc).  The options are in the list and in the list there was two light options. But now, when I hard reset it, I don't see the double light icons anymore.
But the ghost event still happens. I see two events in home assistant, even when I press only ones. The second event usually comes after 2-7 seconds. But it doesn't happen all the time. Sometimes it works fine.

gfwilliams, 2024-04-02:
I'll need that `bthome.json` file mentioned above before I can really figure out what's going on I'm afraid. As-is, it works for me, so I need to see how yours is configured

pinq-, 2024-04-02:
Sure! Here is the button.json file content:
`{buttons:[{name:"light",icon:"light",n:0,v:"press"},{name:"light",icon:"light",n:0,v:"press"},{name:"fan",icon:"fan",n:1,v:"press"}],showBattery:true}`

gfwilliams, 2024-04-02:
Ok, thanks! Please can you give the version 0.03 from the development app loader a try?

I spotted two issues that might fix this:

* You had two identical buttons - due to the way we were putting those in the menu, they just appeared as a single item. That's almost certainly the source of your two identical buttons. Now you'll still get that but you can see them in the app/settings and can remove/change one
* I wasn't handling multiple button numbers quite right, so as a result when you pressed the `light` button it would have created and advertised a third button event, and that could be what was causing you so much trouble.

Now it appears to work ok - I deleted your second `light` (but it shouldn't have mattered) and if you load the `BTHome` app and then paste in `NRF.setAdvertising = print;` in the IDE:

```
>NRF.setAdvertising = print;
=function () { [native code] }
// ------------- I tap 'light' here, it advertises first button pressed
{
  "64722": [ 64, 0, 205, 1, 88, 58, 1, 58, 0 ]
 } { "interval": 50, "whenConnected": true }
// ------------- a few seconds later - advertises no events
{
  "64722": [ 64, 0, 206, 1, 95, 58, 0, 58, 0 ]
 } {  }
// ------------- I tap 'fan' here, it advertises second button pressed
{
  "64722": [ 64, 0, 207, 1, 88, 58, 0, 58, 1 ]
 } { "interval": 50, "whenConnected": true }
// ------------- a few seconds later - advertises no events
{
  "64722": [ 64, 0, 208, 1, 96, 58, 0, 58, 0 ]
 } {  }

// ------------- I tap 'light' here, it advertises first button pressed
{
  "64722": [ 64, 0, 209, 1, 89, 58, 1, 58, 0 ]
 } { "interval": 50, "whenConnected": true }
// ------------- Quickly I tap 'fan' here, it advertises second button pressed
{
  "64722": [ 64, 0, 210, 1, 88, 58, 0, 58, 1 ]
 } { "interval": 50, "whenConnected": true }
// ------------- a few seconds later - advertises no events
{
  "64722": [ 64, 0, 211, 1, 96, 58, 0, 58, 0 ]
 } {  }
```

pinq-, 2024-04-05:
It is still doing it. And every time I try to demo it or film it, it doesn't do it. Can you re open this issue? I will look later with IDE, what is going on and why id dose this randomly. really hard to catch it when it appears occasionally

gfwilliams, 2024-04-09:
Did you change the settings first so there weren't the two identical buttons any more?

Maybe you can try with the `NRF.setAdvertising = print;` line like I did above. It won't advertise anything when you do that so won't work, but at least it'll be obvious what it's outputting and if it's wrong

pinq-, 2024-04-11:
ok, I have maybe idea what's going on.
In our home, we have [shelly plus 1pm](https://kb.shelly.cloud/knowledge-base/shelly-plus-1pm) and they can be as bluetooth proxys. I noticed that the "ghoste" effect only happens in some places. So now I turned off the proxy effect and it looks like it is working. Can't say yet, because it worked randomly. But that might be it. I will let you know soon, did this work.
If it is true, I have two options ( or more?)
1. I turn off the bluetooth option from shelly
2. In home assistant, I only take one(1) event in some time. 
3. Make same code edition in the app. I was trying to look in the home assistant node red, can I identify original signal, but no luck with that.  Maybe I can sent "button down" and "button up" message or running number in watch that will be sent to home assistant.

gfwilliams, 2024-04-11:
> But there is a new problem: if you edit a button and then save it it greats a second one and if you save again, then there is a third one.

This has always been there by the look of it - I'll do a fix.

> we have [shelly plus 1pm](https://kb.shelly.cloud/knowledge-base/shelly-plus-1pm) and they can be as bluetooth proxys

Ok, well, that makes a lot of sense then. It's possible that Shelly's implementation reports advertising data in a different way to other BTHome receivers, so that as you move between rooms home assistant gets different data and thinks the reported values changed.

If you can debug that in Home Assistant I'd report it to Shelly

Maybe try the BTHome app **and leave Button # always at 0 for every action**. You can do different actions by changing 'Action' to press/double-press/etc

It's entirely possible that Shelly has been tested with normal events but forwards >1 event differently.

gfwilliams, 2024-04-11:
Duplicate buttons fixed with https://github.com/espruino/BangleApps/commit/12716a172a7e108be483e13af2f7a4f445627053

And if the rest is Shelly (which seems likely) there's nothing we can do on the app side - it appears to be working as expected

commit 21a17cc7 (2024-04-02): 0.03: Cope with identical duplicate buttons (fix #3260)
      Set 'n' for buttons in Bangle.btHomeData correctly (avoids adding extra buttons on end of advertising)
closed 2024-04-11 by gfwilliams