espruino/BangleApps #3273: [Timer clock / timerclk] countdown setup only increments in seconds after v0.06

Aracnoss, 2024-03-17:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Describe the bug
After updating to v0.06 of Timer Clock, when using the watch faces countdown (swiping left on the watch face), it now only increments in the countdown in seconds, rather than based on where you swiped on the interface.

### Expected behaviour
Swiping in the appropriate place on the interface should increment the appropriate column, i.e. leftmost for hour, middle for minute, rightmost for seconds.

### Steps to reproduce
1. Update to v0.06 of Timer clock
2. Attempt to setup a countdown using minute or hour column.
3. Only seconds column changes regardless of swipe point.

### Installed apps

wid_a_battery_widget (0.04), cprassist (0.02), files (0.08), fileman (0.03), loadingscreen (0.01), chargeanim (0.02), dclock (0.10), widmessages (0.05), ptlaunch (0.15), score (0.01), widbt (0.09), qmsched (0.10), clock_info (0.07), wid_edit (0.05), messages (0.62), rebbleagenda (0.01), oxofocus (0.02), notifyfs (0.12), weather (0.26), alarm (0.46), widdst (0.05), spotrem (0.11), dragboard (0.10), drained (0.04), android (0.33), autoreset (0.01), setting (0.70), messagegui (0.76), widadjust (0.03), about (0.15), sched (0.24), widalarm (0.02), messageicons (0.07), themesetter (0.05), 2047pp (0.04), acmaze (0.04), diceroll (0.02), dtlaunch (0.25), timerclk (0.06), boot (0.60), poweroff (0.04), a_dndtoggle (0.02), lightswitch (0.10), icons (0.02), popconlaunch (0.05)

nxdefiant, 2024-03-17:
I think I see it:
https://github.com/espruino/BangleApps/commit/2e4cb8eb11cfae70c02974c9d7395b25993a1a71#diff-402de28673085438ca8417f9c365b1ac70766e8a6689f297c67c1027ca833bda

dragBorderHrsMins and dragBorderMinsSecs has been commented because they are unused in the current javascript file.
But they seem to be used in the lib:
https://github.com/espruino/BangleApps/blob/master/apps/timerclk/lib.js#L101

Ping @atjn

atjn, 2024-03-17:
Sorry about that! I have a temporary fix ready in #3274 @Aracnoss if possible, can you test that everything works in that version?

Aracnoss, 2024-03-17:
Apologies, I can't test it atm, as I'm away for the weekend, updated via phone. @atjn if you're able to load the GitHub page for your fork then I could give it a quick test, but otherwise I'll have to check in the late evening.

atjn, 2024-03-18:
Fair enough, I will test it sometimes today :)

PR #3274 (2024-03-18): Revert changes to timerclk
Closes #3273

The `timerclk` app uses indirect variable access and it fooled me into believing that this change was safe.
closed 2024-03-18 by thyttan