espruino/BangleApps #3281: [sched] Corrupted alarm when it is snoozed through midnight

elfehr, 2024-03-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Bug description

When an alarm in the evening is repeatedly snoozed until past midnight, its scheduled time becomes invalid and the alarm app is unusable.
- The next alarms do not run, including the snoozed one.
- The alarm app can't be opened. It freezes on "loading…" until I reboot the watch.
- In the app loader, the scheduler app doesn't shows a time for this alarm. Putting the time back and uploading it to the watch solves the problem.
- In sched.json, the alarm changes from `{t:72000000,del:false,on:true,rp:true,as:true,dow:127,last:13,vibrate:"::",msg:"Bedtime",ot:72000000}` to `{t:86400000,del:false,on:true,rp:true,as:false,dow:127,last:13,vibrate:"::",msg:"Bedtime",ot:72000000}`. If I understand correctly, `t:86400000` indicates an invalid time.

### Expected behavior

- An alarm snoozed late in the evening continues to ring after midnight (use case example: a bedtime alarm).
- An invalid alarm doesn't silently prevent the other alarms to ring.
- And ideally, an invalid alarm can be repaired on the watch itself. 

### Steps to reproduce

1. Change the time of the watch to 23:25.
2. Schedule an alarm for 23:30.
3. Snooze the alarm (or let the it auto-snooze) when it rings at 23:30, 23:45 and 00:00.
4. The alarm app won't open anymore.

### Installed apps

widbthide (0.01), chargent (0.05), swiperclocklaunch (0.06), alarm (0.46), iconlaunch (0.19), swp2clk (0.04), notify (0.13), pomoplus (0.06), boot (0.60), sched (0.24), kbmulti (0.08), mylocation (0.11), daisy (0.11), setting (0.71), health (0.29), widalarm (0.02)

nxdefiant, 2024-03-21:
Looks like we need to stay below 86400000. I will take a look unless someone wants to beat me on it.

nxdefiant, 2024-03-21:
I have pushed a possible fix to my repository. Feel free to test the change by updating alarm to 0.47 and sched to 0.25
https://nxdefiant.github.io/BangleApps/index.html?id=alarm
https://nxdefiant.github.io/BangleApps/index.html?id=sched

You will need to resave (or delete) the broken alarm in the alarm app to get it to execute again.

elfehr, 2024-03-22:
That seems to solve it! Thanks.

bruceblore, 2024-03-25:
> I have pushed a possible fix to my repository. Feel free to test the change by updating alarm to 0.47 and sched to 0.25 https://nxdefiant.github.io/BangleApps/index.html?id=alarm [...]

For me, alarms didn't work at all unless I have loaded an app after midnight, and the issue had nothing to do with snoozing. You seem to have fixed that too.

nxdefiant, 2024-03-25:
@bruceblore Feel free to fill a big issue if you find such oddities.

closed 2024-03-25 by gfwilliams

nxdefiant, 2024-03-25:
@elfehr sched and alarm are now in the development app loader, please reinstall from there because I also fixed the timer after you tested.

https://espruino.github.io/BangleApps/?id=sched
https://espruino.github.io/BangleApps/?id=alarm

commit a05216ca (2024-03-25): Merge pull request #3283 from nxdefiant/master

alarm, sched: Fix #3281
commit b135420a (2024-03-25): alarm, sched: Fix #3281

Fix timer
commit db32e9df (2024-03-25): alarm, sched: Fix #3281

Fix wrap around when snoozed through midnight