espruino/BangleApps #3284: [activityreminder] Suddenly "Uncaught Error: Function "getHours" not found" error

Ishidres, 2024-03-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Bug description
Out of nowhere I suddenly get the following error when opening the activity reminder app:

```
Uncaught Error: Function "getHours" not found!
 at line 9 col 18 in activityreminder.app.js
var h = date.getHours();
             ^
in function "getHoursMins" called from line 29 col 67 in activityreminder.app.js
...vityreminder_data.stepsDate), y+=h);
                              ^
in function "drawInfo" called from line 49 col 14 in activityreminder.app.js
    drawInfo();
             ^
in function "run" called from line 56 col 7 in activityreminder.app.js
  run();
      ^
in function called from line 58 col 4 in activityreminder.app.js
})();
   ^
```
Also, I don't get any inactivity alerts anymore when sitting for too long, so I assume this that has to do with each other. 

The weird thing is that this issue started appearing out of nowhere. The last commit in the activityreminder app was done on Feb 23, 2023 and since I first installed the app it worked for many months, then suddenly it stopped. This is why I believe it might be caused/related to another app's boot or widget code.

Unfortunately, I'm clueless idea which app could cause this so I'm opening this issue.

### Installed apps

antonclk (0.11), widlock (0.08), notify (0.13), quicklaunch (0.15), authentiwatch (0.07), launch (0.20), slevel (0.04), weather (0.26), thermom (0.07), bee (0.03), intervalTimer (0.01), pomoplus (0.06), noteify (0.03), widbatv (0.03), info (0.03), barometer (0.04), torch (0.11), iconlaunch (0.19), sportmode (0.01), scicalc (0.03), qrcode (0.06), qmsched (0.10), alarm (0.46), messagesmusic (0.05), messages (0.62), imgclock (0.11), boot (0.60), widalarm (0.02), messageicons (0.07), hrm (0.12), circlesclock (0.26), dtlaunch (0.25), agenda (0.15), run (0.18), kbmulti (0.08), sched (0.24), 2047pp (0.04), android (0.34), recorder (0.39), icons (0.02), inspire (0.03), stopwatch (0.06), smpltmr (0.09), clock_info (0.08), widmessages (0.06), mylocation (0.11), setting (0.71), clkinfosunrise (0.05), edisonsball (0.04), sleepphasealarm (0.18), messagegui (0.77), health (0.29), sleeplog (0.16), healthscore (0.01), widhr (0.01), widclk (0.08), widbthide (0.01)

thyttan, 2024-03-21:
Did you first get this bug now, or have you noticed it for some time?

Ishidres, 2024-03-21:
Yes, I've noticed it for some time (~ 1-2 months) but at first I just thought I messed up the app settings somehow and I wouldn't get any alerts because of that until I eventually connected it to the IDE and noticed the error logs.

nxdefiant, 2024-03-22:
Can you please share your activityreminder.data.json? I took a small look and it might be related to espruino/Espruino@805c317
Also see https://forum.espruino.com/comments/17322019/

Ishidres, 2024-03-22:
Certainly, here it is:
```
{firstLoad:false,stepsDate:{ms:1711107487213.85571289062},stepsOnDate:2757,okDate:{ms:1970},dismissDate:{ms:1970},pauseDate:{ms:1970}}
```
However, I already tried reinstalling the app along with deleting the activityreminder.data.json after the problem occured so the JSON above was generated after a reinstallation.

nxdefiant, 2024-03-22:
I suspect it is the date format {ms:1711107487213.85571289062} in the current firmware. What you could try is deleting the json after updateing the firmware to bleeding edge.

Ishidres, 2024-03-22:
> What you could try is deleting the json after updateing the firmware to bleeding edge.

Sorry, what do you mean with updating the firmware to bleeding edge? I use the latest firmware version and I use a custom apploader where I always fetch the latest commits from master, so I think I'm already using the newest firmware version?

thyttan, 2024-03-22:
Current cutting edge is fw2v21.63. You can find the latest cutting edge fw e.g. after all stable ones on the fw updater interface on the app loader.

Ishidres, 2024-03-22:
Before updating the fw, we could try editing the lib code to this:
```diff
diff --git a/apps/activityreminder/lib.js b/apps/activityreminder/lib.js
index a5c35190c..a6bb155b3 100644
--- a/apps/activityreminder/lib.js
+++ b/apps/activityreminder/lib.js
@@ -31,14 +31,14 @@ exports.loadData = function () {
   },
     require("Storage").readJSON("activityreminder.data.json") || {});
 
-  if (typeof (data.stepsDate) == "string")
-    data.stepsDate = new Date(data.stepsDate);
-  if (typeof (data.okDate) == "string")
-    data.okDate = new Date(data.okDate);
-  if (typeof (data.dismissDate) == "string")
-    data.dismissDate = new Date(data.dismissDate);
-  if (typeof (data.pauseDate) == "string")
-    data.pauseDate = new Date(data.pauseDate);
+  if (typeof (data.stepsDate) == "number")
+    data.stepsDate = new Date(data.stepsDate * 1000);
+  if (typeof (data.okDate) == "number")
+    data.okDate = new Date(data.okDate * 1000);
+  if (typeof (data.dismissDate) == "number")
+    data.dismissDate = new Date(data.dismissDate * 1000);
+  if (typeof (data.pauseDate) == "number")
+    data.pauseDate = new Date(data.pauseDate * 1000);
 
   return data;
 };
```
However, I'm not sure if this will break the app for users who did not update the firmware to the latest stable version, yet. Any idea?

nxdefiant, 2024-03-22:
You could do the same that I have done in PR #3277 for another app.

Basically
`new Date(typeof data.okDate === 'string' ? data.okDate : data.okDate.ms)`

another option would be

`new Date(data.okDate.ms ? data.okDate.ms : data.okDate)`

Ishidres, 2024-03-22:
Possible hotfix in https://github.com/espruino/BangleApps/pull/3286
I'm currently testing it.

Ishidres, 2024-03-22:
Looks like it's working again.

closed 2024-03-25 by bobrippling

zerodogg, 2024-04-03:
Time to tag a new release for this?

nxdefiant, 2024-04-03:
Good question
@bobrippling Why was there no update to changelog/metadata?

thyttan, 2024-04-03:
I just bumped the version :ok_hand:

Ishidres, 2024-04-04:
Thanks everyone! :)