espruino/BangleApps #3289: Lock event not declared in Typescript

paul-arg, 2024-03-23:
Hello,

I am trying to use `Bangle.on('lock', function (on, reason) {});` in a Typescript file, but the `lock` event does not seem to be declared in Typescript.

This error is raised: 

```
No overload matches this call.
  The last overload gave the following error.
    Argument of type '"lock"' is not assignable to parameter of type '"midnight"'.ts(2769)
main.d.ts(3747, 10): The last overload is declared here.
```

Might there be some more events not yet ported to Typescript?

Thanks

gfwilliams, 2024-03-25:
I'm not sure what's wrong, because it's definitely defined in https://github.com/espruino/BangleApps/blob/master/typescript/types/main.d.ts#L3630

paul-arg, 2024-03-25:
Ah, I think TS is not happy because I am using the arguments `on` and `reason`.
If I just call `Bangle.on("lock", function (on: boolean) {});` without `reason` the error disappears.
It looks like `reason` was not added to the TS declaration.

gfwilliams, 2024-03-25:
Looking at it, as far as I can tell I think the `lock` typescript is automatically created, so it's just that `reason` was added semi-recently (6 months?) and the typescript decls haven't been updated since then

closed 2024-03-25 by halemmerich