espruino/BangleApps #3293: [banglejs gadgetbridge] message REPLY not handled?

Stuff-etc, 2024-03-24:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Intro
I wanted to add some reply options as mentioned here: https://forum.espruino.com/conversations/373668/ but GB doesn't seem to use them.

### How to reproduce:
* Add a `console.log("E:",event)` to line 22 of `android.boot.js`
* wait for an incoming message:
```
E: {
  "t": "notify",
  "id": 1711259714,
  "src": "WhatsApp",
  "title": "Test Person",
  "subject": "",
  "body": "hi!",
  "sender": ""
 }
```
* manually answer: `Bluetooth.println(GB({"t":"notify","id":1711259714,"n":"REPLY","msg":"test response"}))`
```
E: {
  "t": "notify",
  "id": 1711259714,
  "n": "REPLY",
  "msg": "test response"
 }
```
* **nothing happens**

gfwilliams, 2024-03-25:
I think it depends on the app that's being used - not all apps offer reply functionality straight from the notification. I guess WhatsApp might not be one that does?

Stuff-etc, 2024-03-26:
I don't think that this is the case here. 
- I can answer from the notification straight on my phone.
- I didn't test it recently, but it worked with GB and Pebble[ 2 years ago](https://forum.espruino.com/conversations/373668/).

I think there is a problem in the BangleJS GB. Do you know if the reply-function was ever successfully tested? If so: can you recommend an app where this should work?
Also: Is this the right place for this issue or is there a BJS-GB repo?

gfwilliams, 2024-03-27:
> Do you know if the reply-function was ever successfully tested?

Yes - I added the code roughly 2 years ago (June 2022) and I'm pretty convinced it worked fine at that point.

However looking back at the code, the implementation is quite involved ([see here](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L648-L654)) so it wouldn't be a total surprise if something got broken.

I've just added a fix that tidies things up and hopefully it'll be merged soon: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3652

But yes, for Gadgetbridge issues it's best to report them in Gadgetbridge itself https://codeberg.org/Freeyourgadget/Gadgetbridge/issues

closed 2024-03-27 by gfwilliams

Stuff-etc, 2024-03-27:
thanks, appreciated! Will check it after my vacation!