espruino/BangleApps #3301: [Recorder] Feature suggestion: wait for GPS fix

thyttan, 2024-03-26:
Should we implement some optional wait mechanism for getting a GPS fix before starting recording where it's included?

As an example the `run` app starts looking for a fix on launch and indicates to the user when a fix is established.

Maybe as soon as the GPS box is ticked `recorder` should start trying to establish a fix? If the user has "GPS Widget" installed they will see when they have a fix.

Or there could be an question presented when ticking the `RECORD` box asking if it should wait for a fix or not.

This issue prompted me thinking about this: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/3648

Thoughts on this are appreciated :)

nxdefiant, 2024-03-27:
> Should we implement some optional wait mechanism for getting a GPS fix before starting recording where it's included?

I would say no. Sometimes the watch needs a little longer to get a fix, so lets record the values you have, you can always filter later the records without GPS data. The only exception would be if gps is the only thing to get logged.

> Maybe as soon as the Log GPS box is ticked recorder should start trying to establish a fix. 

Due to the heavy power draw of the GPS I would say no. Maybe the user wants intentionally only partial GPS log?
Or we could add a checkbox "enable GPS" the user can decide if recorder should enable it or not.

halemmerich, 2024-03-27:
> Should we implement some optional wait mechanism for getting a GPS fix before starting recording where it's included?

Optionally waiting would be OK but I think it should be easy enough to do that in post processing in GB. Discarding all entries until the first one with a location should do the trick.

> Maybe as soon as the Log GPS box is ticked recorder should start trying to establish a fix.

That is only relevant if in the Recorder app and doing the first toggle. On my watch "Log GPS" is always on and I practically do never start recording using the Recorder app. So turning GPS on in the Recorder App for a quicker fix would be useless for me since I only open it to look at existing tracks.

thyttan, 2024-03-27:
Maybe it's best if I just add a note re this on `gadgetbridge.md` in EspruinoDocs.

bobrippling, 2024-03-27:
I agree - I can't remember but if we log a record with a way of indicating that there's no GPS fix, I think that's fine and later tools can filter out those records if needed

thyttan, 2024-03-27:
Thanks for chiming in all three 🛎️🙂

closed 2024-03-27 by thyttan