espruino/BangleApps #3310: [Bordle] Bordle is only playable once

PVTejas, 2024-03-29:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

When Bordle is installed after a factory reset and the game is completed once (only tested with games ending in a successful guess so far), it quits to the watch face. Attempting to open the game again just makes it get stuck on a "Loading" dialog box.

### Installed apps

messagegui (0.77), android (0.34), boot (0.60), antonclk (0.11), notify (0.13), health (0.29), setting (0.71), about (0.15), widalarm (0.02), sched (0.25), alarm (0.47), widbt (0.09), widid (0.03), widlock (0.08), puzzle15 (0.05), authentiwatch (0.07), stacker (0.03), wid_a_battery_widget (0.04), f9lander (0.04), invader (0.12), astroid (0.06), numberchaser (0.01), multidice (1.30), trex (0.05), tetris (0.12), waternet (0.03), sokoban (0.03), chess (0.05), acmaze (0.05), taglaunch (0.04), mmind (0.02), snaky (0.01), bordle (0.03), sleeplog (0.16), widdevst (0.03), widclk (0.08), messageicons (0.07), hrm (0.12), widmessages (0.06), messages (0.62), widpedom (0.22)

bobrippling, 2024-04-02:
@berkenbu is this something you're familiar with? @PVTejas it could be a bug, are you able to connect [the IDE](https://www.espruino.com/ide/) and see any logs?

PVTejas, 2024-04-11:
[image]

apologies for the late response. Here is the error I see in the IDE.

berkenbu, 2024-04-11:
That looks like something went wrong with reading the statistics file, although the printout looks fine. Could you try to delete the file 'bordlestats.json' via the web IDE and see if that helps?

PVTejas, 2024-04-12:
Managed to delete `bordlestats.json` via the web IDE using `require("Storage").erase("bordlestats.json")`.

Bordle then launches, and once I successfully guess the word (or cheat from the console), I have the same issue again.

The `Got INT expected '}'` confuses me... why would it expect a closing brace? Are INTs not valid dict keys in js?

Can we compare with your bordlestats.json just to make sure the format's the same?

halemmerich, 2024-04-12:
Numbers are valid as keys in JS but are not valid as keys (or names as the RFC calls them) in JSON. I believe bordle does nothing wrong here.
#3346 could be the same problem as well.

Espruino had a change for `writeJSON/readJSON` between 2v19 and 2v20 that does not write the quotation marks to save space. [https://github.com/espruino/Espruino/commit/59dda13c259888671276a55dbcdb7579223d6329](https://github.com/espruino/Espruino/commit/59dda13c259888671276a55dbcdb7579223d6329).

There were several changes after that to JSON writing/parsing, so maybe @gfwilliams got a clue as to what is happening here?

This does fail exactly like bordle does on 2v21:
```
require("Storage").writeJSON("test",{"1":1});
require("Storage").readJSON("test");
```

berkenbu, 2024-04-12:
Ah, thank you for figuring this out, this seems to indeed be the issue.
I could change Bordle to not use integers as keys, but I feel this should really be fixed at the firmware level.

PVTejas, 2024-04-14:
`require("Storage").write("bordlestats.json","{'1':0,'2':0,'3':0,'4':0,'5':0,'6':0,'p':0,'w':0,'s':0,'ms':0}")`

Running this and attempting to read it works as expected, but every time bordle updates `bordlestats.json`, it gets broken again

PVTejas, 2024-04-15:
So I managed to fix it on my device by going into the IDE and modding `bordle.app.js`.

As @berkenbu said, this really should be fixed at the firmware level so I won't be committing any of these changes to the repo, but if anyone wants the app functional in the meantime here are the instructions:

1. Go to www.espruino.com/ide, connect to your device using the button in the top left
2. Access device storage (the button with 4 disks) and save `bordle.app.js` to your computer
3. Open bordle.app.js on your computer and replace all instances of `require("Storage").writeJSON(filename,data)` with `require("Storage").write(filename,JSON.stringify(data))`. Don't forget to save
4. Click the storage button in the Web IDE again and select upload file
5. Upload the new version of `bordle.app.js`
6. If `bordlestats.json` already exists, you might want to either modify that to have quotes around the dict keys or straight up delete it and let the modified app create one from scratch.

PVTejas, 2024-04-15:
Upstream source code fixed at https://github.com/espruino/Espruino/commit/b251d8314ff01d40f6e451648c5c5907f54fdf0b

closed 2024-04-15 by PVTejas