espruino/BangleApps #3332: sleeplog: Suddenly almost always logged as awake on stable and development branch

Ishidres, 2024-04-04:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Describe the bug
Since around March 28 I notice that the sleep reported by sleeplog is almost always reported as "awake" but never asleep for a long consecutive time (see screenshot).

[image]
- In the screenshot above I was not wearing the Bangle.JS from 19:40 to 21:40, however it only marked the beginning and end of that period as "not worn".
- The time from around 01:00 to 09:00 should be marked as sleep or consecutive sleep, however it's marked as almost only awake, though I did wear the band during that time.

### Steps to reproduce

- I should add that for testing I installed the stable branch sleeplog version for the duration from 01:00 to 09:00 and the development version (that also contains the latest temperature threshold wear detection features) after I woke up at around 09:00.
- I reset the thresholds to the default values and set the temperature wear threshold to 29°C.
- I did a full reinstall of sleeplog several times where I also removed all files generated by sleeplog (including source files as well as recorded data)

I first suspected it might be due to the latest temperature checks that were added but it also happens on the stable branch. I also deleted some apps that I suspected might potentially cause issues like `activityreminder` but this changed nothing.

I'm currently using my own fork but as you can see in https://github.com/espruino/BangleApps/compare/master...Ishidres:BangleApps:master there are no significant changes that should cause any problems with sleeplog.

### Installed apps

widlock (0.08)
notify (0.13)
quicklaunch (0.15)
authentiwatch (0.07)
slevel (0.04)
weather (0.26)
thermom (0.07)
bee (0.03)
intervalTimer (0.01)
pomoplus (0.06)
noteify (0.03)
widbatv (0.03)
info (0.03)
barometer (0.04)
torch (0.11)
iconlaunch (0.19)
scicalc (0.03)
qmsched (0.10)
messagesmusic (0.05)
messages (0.62)
boot (0.60)
widalarm (0.02)
messageicons (0.07)
circlesclock (0.26)
agenda (0.15)
run (0.18)
kbmulti (0.08)
2047pp (0.04)
android (0.34)
icons (0.02)
inspire (0.03)
stopwatch (0.06)
smpltmr (0.09)
widmessages (0.06)
mylocation (0.11)
setting (0.71)
clkinfosunrise (0.05)
edisonsball (0.04)
sleepphasealarm (0.18)
widhr (0.01)
widclk (0.08)
widbthide (0.01)
clock_info (0.09)
sched (0.25)
alarm (0.47)
health (0.30)
hrm (0.13)
recorder (0.42)
messagegui (0.78)
sleeplog (0.17)

bobrippling, 2024-04-05:
Yeah sounds like a regression by either #3317 or #3280, probably like you say, the wear/temperature one since the other was only merged recently. We could narrow it down by reverting to the version before the wear/temperature one - also do you see any console warnings in [the IDE](https://www.espruino.com/ide/)?

Ishidres, 2024-04-06:
Good idea, thanks. I think it is related to #3280 because this night I tried the stable version of sleeplog without temperature checks again and it worked. I think it's good to first start reverting #3280 only and not the other PR to narrow it down.

thinkpoop, 2024-04-07:
Just as an extra data point: I've been using the SleepLog dev versions without any issues. I set it to 27 C the first day and haven't thought about it since.

Ishidres, 2024-04-07:
Yes it's really strange. I'm now using the stable version of sleeplog from https://banglejs.com/apps/ and I get the same issues that I reported above. That makes me guess that it's unrelated to any of the recent code changes and might be an issue like in #3284 where the time format suddenly changed and the app stopped working. Maybe it's the same issue in sleeplog again and e.g. a firmware update caused new date formats.

I could try upgrading to the cutting edge firmware but I feel like I'll break much more with that.

thyttan, 2024-04-07:
> I could try upgrading to the cutting edge firmware but I feel like I'll break much more with that.

That would probably be the quickest way to verify if it's the date thing. Cutting edge is stable the majority of the time - I virtually always run it. You can always downgrade to stable 2v21 again :)

Ishidres, 2024-04-07:
Thanks for the info and great, I'll soon do that and report my findings here.

thinkpoop, 2024-04-08:
I had been on bleeding edge firmware and last night downgraded to 2v21. 

It recorded that I was awake at 5am for an hour, which I don't remember. But otherwise it did record sleep the rest of the time I was asleep.

Ishidres, 2024-04-08:
Have you enabled debugging and checked that maybe the thresholds are high enough? Maybe it considered some movement as awake.

thinkpoop, 2024-04-08:
Yeah, it was probably motion. I'm not worried about it, just mentioned since it was an oddity. But otherwise it seems to be working as usual for me.

Ishidres, 2024-04-12:
Update: It seems to be related to another installed app, because I uninstalled a bunch of apps and now it seems to be working again (with higher thresholds). I'll try to figure out which app caused this.

Ishidres, 2024-04-14:
I think this was due to changes I made in the health app in my own fork at https://github.com/espruino/BangleApps/compare/master...Ishidres:BangleApps:master#diff-f18170cb5895182061e4aef64b18f44328fc7ee1f84eedc3fdb623d82e1a8587. I replaced health with the stable app loader version and it seems to work, so I think this issue can be closed. Thanks for the help, guys.

closed 2024-04-14 by Ishidres