espruino/BangleApps #3346: [ptlaunch] writeJSON truncating patterns causing load failures

WhelanB, 2024-04-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

**Describe the bug:**
When creating a pattern in the pattern launcher that does not start in the top left circle, all patterns will disappear from the "Manage Pattern" page and triggering a write to storage will wipe any existing patterns

**Expected Behaviour**
Any pattern sequence created in the pattern launcher should save correctly, and be visible from the "Manage Patterns" page

**Steps to Reproduce**

1. Open Pattern Launcher
2. Click "Add Pattern"
3. Add a pattern that starts in the upper left circle
4. Click "Manage Patterns"
5. Observe the new pattern in the list of patterns
6. Click "Back"
7. Click "Add Pattern"
8. Add a pattern which starts in any circle _but_ the upper left circle
9. Click "Manage Patterns"
10. Observe the pattern list is empty
11. Connect Bangle to Web IDE
12. Open storage and observe patterns are present in ptlaunch.patterns.json
13. In Pattern Launcher, click "Settings"
14. Enable/Disable Lock
15. In the IDE, open storage and observe patterns in ptlaunch.patterns.json are missing

-----------------------------

This seems to be an issue with storage.writeJSON, though I'm not sure if it's expected behaviour and Pattern Launcher is wrong, or something has changed on the Bangle side. The patterns in pattern launcher are stored in JSON as a numeric sequence, with each number being one of the circles and the order being the order of the pattern.

When a pattern starts on the top left circle, the sequence begins with "0", and so is serialised as a string:

```
"012":{app:{name:"Messages",src:"messagegui.app.js"}}
```

however in all other cases, the leading zero will not be present and so the sequence is serialised as an integer, causing pattern launcher to fail to parse it:

```
47:{app:{name:"Android",src:"android.app.js"}}
```

I've been able to work around it by replacing the call to writeJSON with
```
storage.write("ptlaunch.patterns.json", JSON.stringify(data));
```

and am happy to submit this as a fix, but wanted to check that this isn't something that has changed on the environment side first

Thanks!

nxdefiant, 2024-04-09:
Lets bring in the authors @crazysaem @hughbarney

PVTejas, 2024-04-15:
Bug fixed in firmware at https://github.com/espruino/Espruino/commit/b251d8314ff01d40f6e451648c5c5907f54fdf0b, I suppose this can be closed?

bobrippling, 2024-04-17:
Yes I think so - @WhelanB please re-open if you see the issue on the latest firmware

closed 2024-04-17 by bobrippling