espruino/BangleApps #3355: [quicklaunch] Wrongly starts app when swiping clock info

Spioune, 2024-04-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

### Bug

On a clock with 2 (or more?) clock info (like Pebble++ Clock):
1. Select the top right clock info
2. Select the top left clock info
3. Swipe left/right/up/down with the intention to switch the top left clock info will trigger quicklaunch instead.

Based on my tests this only happens on the top left clock info and this only happens if you select the top right clock info prior.

### Steps to reproduce

1. Install a clock with 2 clock info like [Pebble++ Clock](https://banglejs.com/apps/?id=pebblepp)
2. Install [quicklaunch](https://banglejs.com/apps/?id=quicklaunch)
3. In Settings > Apps > Quick Launch, setup a quick action on swipe left, right, up or down
4. Start Pebble++ clock from the main menu
5. Select the top right clock info
6. Select the top left clock info
7. Swipe left, right, up or down
8. The quicklaunch is triggered

This bug seems to happen on every clock with more than 2 clock info (#3347)

### Installed apps

alarm (0.46), widalarm (0.01), sched (0.22), setting (0.68), health (0.28), about (0.14), launch (0.20), boot (0.59), antonclk (0.11), widlock (0.08), widbat (0.12), widbt (0.09), widid (0.03), welcome (0.14), notify (0.13), quicklaunch (0.15), clock_info (0.09), pebblepp (0.05)

bobrippling, 2024-04-17:
I'd expect this code to cover it:

https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/quicklaunch/boot.js#L36-L36

Perhaps the clkinfo is defocussed and then this handler runs? But this should be covered by this code:

https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/clock_info/lib.js#L286-L287

thyttan, 2024-04-17:
Is it maybe a case of using `.on(...) and not `.prependListener(...)` somewhere? I wouldn't expect that to be it - but maybe?

thyttan, 2024-04-17:
Ok - I think what happens is (these are guesses at the moment)
1. `quicklauch`'s swipe handler is registered before the clock_info's on the clock face so the `stopEventPropagation` hasn't been called yet so `quicklaunch` runs. 
2. This doesn't matter if `CLOCKINFO_FOCUS` is `true`, but somehow when directly clicking the other clock_info when one was active I guess `CLOCKINFO_FOCUS`  first set to `true` once again AND THEN the first clock_info is unfocused setting `CLOCKINFO_FOCUS` `false`. So `quicklaunch` goes ahead and launches an app.

If this is right we have two bugs that can be fixed. (1) by switching from `Bangle.on(...)` to `Bangle.prependListener??Bangle.prependListener(...) : Bangle.on(...)`. (2) can probably be solved in a couple of ways.

thyttan, 2024-04-17:
> If this is right we have two bugs that can be fixed. (1) by switching from `Bangle.on(...)` to ` if (Bangle.prependListener) {Bangle.prependListener("swipe",swipeHandler)} else {Bangle.on("swipe",swipeHandler)};` in clock_info library.

This change seems to help some. But I get errors after fidgeting back and forth with the clock infos on `pebblepp`. E.g. :

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v21.84 (c) 2024 G.Williams
Uncaught Error: Can't read property 'on' of undefined
 at line 242 col 8 in clock_info
    itm.on('redraw', options.redrawHandler);
       ^
in function "menuShowItem" called from line 284 col 60 in clock_info
...menuA].items[options.menuB]);
                              ^
in function called from system
Uncaught Error: Can't read property 'blur' of undefined
 at line 295 col 12 in clock_info
    if (itm.blur && itm.blur(options) === false)
           ^
in function "blur" called from line 330 col 16 in clock_info
          blur();
               ^
in function called from system
Uncaught Error: Can't read property 'focus' of undefined
 at line 305 col 14 in clock_info
      if (itm.focus && itm.focus(options) === false)
             ^
in function "focus" called from line 320 col 15 in clock_info
        focus();
              ^
in function called from system
> 
```

thyttan, 2024-04-17:
I did two rounds of tests to find out how the `Bangle.CLKINFO_FOCUS` behaves on `pebblepp``:

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v21.84 (c) 2024 G.Williams
>// Before any touch interaction:
=undefined
>Bangle.CLKINFO_FOCUS
=undefined
>// After clicking left clock info:
=undefined
>Bangle.CLKINFO_FOCUS
=true
>// After clicking right clock info:
=undefined
>Bangle.CLKINFO_FOCUS
=true
>// After clicking left clock info again:
=undefined
>Bangle.CLKINFO_FOCUS
=undefined
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v21.84 (c) 2024 G.Williams
>// Before any touch interaction:
=undefined
>Bangle.CLKINFO_FOCUS
=undefined
>// After clicking right clock info:
=undefined
>Bangle.CLKINFO_FOCUS
=true
>// After clicking left clock info:
=undefined
>Bangle.CLKINFO_FOCUS
=undefined
>// After clicking right clock info again:
=undefined
>Bangle.CLKINFO_FOCUS
=true
> 
```

So it looks like first selecting the left one and then the right one works for keeping `CLKINFO_FOCUS` `true`. But not the other way around.

bobrippling, 2024-04-18:
Yes, I think bug 2 is the more crucial one here / might solve the problem if we can fix it. I think you're right - after debugging I've found that `CLKINFO_FOCUS` will be set (to `true`) by the first handler (left hand clkinfo) and then unset (/ `delete`d) by the second handler (right hand clkinfo).

This is done in [the `touchHandler`](https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/clock_info/lib.js#L313-L324).

So while #3364 will help, I believe we need to deal with two clkinfos interfering with each other. There are two ways I can think of doing this:

CLKINFO_FOCUS becomes a count

- So when we have a tap on a previous clkinfo (previous as in, earlier in the event handler list), it'll focus itself, `CLKINFO_FOCUS++`
- The later handler comes along, unfocuses itself, `CLKINFO_FOCUS--`
- `CLKINFO_FOCUS` is now `1`, so clkinfo is registered as being focussed still, meaning quicklaunch will work because of:

> https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/quicklaunch/boot.js#L36-L36

Alternatively, we cycle through other clkinfos when defocussing, attempt to see if they'll focus themselves and so don't under `CLKINFO_FOCUS`, this feels more complicated, error-prone and not the responsibility of a single clkinfo, so I vote the former. I've pushed up a PR, let me know what you think

bobrippling, 2024-04-18:
Thanks! Yes I think [this is the discussion](https://github.com/espruino/BangleApps/pull/3234#discussion_r1520984877), happy to continue with this fix for now and we can always drop it after `stopEventPropagation`'s been out for some time

PR #3366 (2024-04-19): clkinfo: make `CLKINFO_FOCUS` a counter
This avoids issues when we have multiple clkinfos visible. The first clkinfo may want to focus itself, but a second clkinfo may then unfocus, deleting the first's effects (i.e. setting `CLKINFO_FOCUS`).

In detail:

`CLKINFO_FOCUS` will be set (to `true`) by the first handler (left hand clkinfo) and then unset (/ `delete`d) by the second handler (right hand clkinfo).

This is done in [the `touchHandler`](https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/clock_info/lib.js#L313-L324).

This commit makes `CLKINFO_FOCU [... cut]
commit 2bbef9f7 (2024-04-19): clkinfo: make `CLKINFO_FOCUS` a counter

This avoids issues when we have multiple clkinfos visible. The first clkinfo may want to focus itself, but a second clkinfo may then unfocus, deleting the first's effects (i.e. setting `CLKINFO_FOCUS`).

In detail:

`CLKINFO_FOCUS` will be set (to `true`) by the first handler (left hand clkinfo) and then unset (/ `delete`d) by the second handler (right hand clkinfo).

This is done in [the `touchHandler`](https://github.com/espruino/BangleApps/blob/c2ea454a3bc7bfaefbc39a5b376510bb9d7d4ec8/apps/clock_info/lib.js#L313-L324).

This commit makes `CLKINFO_FOC [... cut]
closed 2024-04-19 by thyttan