espruino/BangleApps #3429: [Pebble++ Clock/clkinfocal] Clock Info - Calendar format dd MMM shows day twice

vczb, 2024-05-20:
### Affected hardware version

Bangle 2

### Your firmware version

BangleJS 2 - 2v21

### The bug

[image]

In the "Pebble++ Clock" with "Calendar Clockinfo," when using the format `dd MMM`, the day appears twice. First, you can see the day at the top inside the icon, then below again along with the month.

I suggest removing the day from the bottom to show just:

```
20
May
```

Another option is to add a new date format, perhaps `dd MM` (or something similar).

---

App code:

https://github.com/espruino/BangleApps/blob/master/apps/clkinfocal/clkinfo.js

---

If this issue makes sense, I would love to work on it.

Thanks a lot.

### Installed apps

- All default apps except Anton Clock
- Pebble++ Clock
- Calendar Clockinfo

bobrippling, 2024-05-21:
Sounds good, I don't think @hughbarney you'd have any objections?

vczb, 2024-05-22:
Since there are no objections I will start to work on it

PR #3432 (2024-05-23): clkinfocal: remove duplicate day in calendar when format setting is 'dd MMM'
Remove duplicated day in calendar when format setting is 'dd MMM'

[image]

You can test the branch here https://vczb.github.io/BangleApps/?c=clkinfo&q=clkinfocal

Close #3429
closed 2024-05-23 by bobrippling