espruino/BangleApps #3433: Touchscreen Not Responding

joshaspinall, 2024-05-23:
### Affected hardware version

Bangle 2

### Your firmware version

2v22

### The bug

The touchscreen of my device stops working, seemingly at random. Not affected by installing or removing apps. Full reset does not help. No touch even on the bootloeader/testing menu when holding down the button. 

Function seemingly returns, but with no apparent pattern. Screen is physically undamaged, no cracks or scratches and is clean. 

### Installed apps

messagegui (0.79)
android (0.35)
boot (0.61)
launch (0.21)
antonclk (0.11)
notify (0.14)
health (0.30)
setting (0.71)
about (0.15)
sched (0.26)
widalarm (0.02)
alarm (0.47)
widbat (0.12)
widbt (0.09)
widlock (0.08)
widid (0.03)
messageicons (0.07)
widmessages (0.06)
messages (0.62)

bobrippling, 2024-05-23:
@gfwilliams, seen anything like this before?

gfwilliams, 2024-05-24:
I haven't really, no... @joshaspinall have you been near any water with it recently? I have seen some touchscreen issues on watches where water has got inside, and if it's still working on and off, it might be worth trying to keep it somewhere warm and dry for a few days and see if that improves matters?

joshaspinall, 2024-05-25:
The watch has never been submerged, but may have had the occasional splash from the rain or washing hands as it is my daily wear. 

I'll give it chance to dry out properly and see how it fares. Thanks for the response.

joshaspinall, 2024-08-12:
Hey @gfwilliams just checking back in on this issue.
It seems the touch problems have only gotten worse, where the touch will not work more often than it does.  I have recently noticed a somewhat significant drop in battery life (2-3 weeks down to 5-7 maximum with little to no active use).
Before I attempt to break open the watch on my hotplate, is there anything you would want me to check out or try?
As before, the watch has continued to be used as a daily wear.  Hasn't gotten wet and the screen is clean and mark free when testing.  Apps unchanged from original report.