espruino/BangleApps #3434: cscsensor 0.11: Cadence is only 0 or 1

nxdefiant, 2024-05-27:
On Bangle.js 2 after updating cscsensor from 0.09 to 0.11 with unchanged firmware 2v21 the value of the cadence sensor is only 0 or 1. It should display > 60 RPM and it did with earlier versions.

Also when connecting to the sensor I get a lot of connection timeouts until the watch finally connects.

bobrippling, 2024-05-27:
@gfwilliams looks like one of these has probably introduced this:

799da731297d48d03f149bbf9a10389059494812
9841aeddebadd501d68bb77068ccfca1e58d6920
5c89b45722e1389137a160ae75224f719440371e

gfwilliams, 2024-06-03:
It'd be really interesting if you could run `csc.on("data", print)` and see if the `crps` value is actually right - whether it's some issue with the library or the interface between the lib and the app.

Or is there any chance of getting a dump of the notification value from your sensor? I tested with a coospo one here and it appeared to work ok for me and report the correct data.

It might also be interesting to try it at this version (before I removed the duplicate libraries) and see if that broke it? https://github.com/espruino/BangleApps/commit/799da731297d48d03f149bbf9a10389059494812

Both of the two forked versions of the CSC sensor library seemed broken in their own separate ways so this *should* have been an improvement overall.

> Also when connecting to the sensor I get a lot of connection timeouts until the watch finally connects.

Did you not get that before as well though? The actual connection code hasn't changed

nxdefiant, 2024-06-03:
You can find the sensor as wahoo cadence sensor btw.

output from `csc.on("data", print)`:
> { "ccr": 101, "lcet": 28099, "crps": 1.20754716981, "cdt": 0.828125 }
{ "ccr": 103, "lcet": 29827, "crps": 1.18518518518, "cdt": 1.6875 }
{ "ccr": 103, "lcet": 29827, "crps": 0, "cdt": 0 }
{ "ccr": 104, "lcet": 30715, "crps": 1.15315315315, "cdt": 0.8671875 }
{ "ccr": 105, "lcet": 31623, "crps": 1.12775330396, "cdt": 0.88671875 }
{ "ccr": 106, "lcet": 32573, "crps": 1.07789473684, "cdt": 0.927734375 }

I think I see it now:
1. cscsensor did multiply the value by 60: https://github.com/espruino/BangleApps/blob/708c295853bfd3c6fbb4549ec2f1878e0b87d190/apps/cscsensor/cscsensor.app.js#L168
2. The check for time change is missing:
https://github.com/espruino/BangleApps/blob/708c295853bfd3c6fbb4549ec2f1878e0b87d190/apps/cscsensor/cscsensor.app.js#L167

> Did you not get that before as well though? The actual connection code hasn't changed

Sometimes, but not that extreme. The only difference I see is the maxInterval setting: https://github.com/espruino/BangleApps/blob/708c295853bfd3c6fbb4549ec2f1878e0b87d190/apps/cscsensor/cscsensor.app.js#L229

gfwilliams, 2024-06-03:
Ahh, ok, great! That should be an easy fix then.

> The only difference I see is the maxInterval setting

Yes, I took that out because it does absolutely nothing at all :) I checked in the firmware and everything - if it were to do anything at all it should be in the call to `.connect`

closed 2024-06-04 by gfwilliams

nxdefiant, 2024-06-12:
Just tested, still not optimal. I have slowly but continously turned the pedal at around 60 RPM, the app displayed either 60 or 0. With the old behaviour it stayed at 60 which I prefer.

The watch was connected to a PC to update the app, but then disconnected. This time I was lucky and got only one connection timeout (usally a few dozen times before it connects).

gfwilliams, 2024-06-12:
Ok, just tried again and I can see the issue - i'll get a fix in soon

gfwilliams, 2024-06-12:
Ok, just pushed a fix now - hopefully it'll be a lot better

gfwilliams, 2024-06-12:
One more remark:

When I'm testing here, my coospo sensor disconnects after a minute of not being used and is unconnectable - so CSCSensor keeps saying it's trying to connect until the wheel or crank has actually been moving. Could that be the issue you're seeing?

nxdefiant, 2024-07-07:
Sorry only found time to test today. Yeah the values do look good now, thanks, but the connection problem still exists.

It seems it is the other way around for me: I can not connect while the cranks are moving, I get the connection timeout message in this case, so the only way it connects is when I start the csc app before turning the sensor on (turning the crank)

gfwilliams, 2024-07-08:
Glad that fixed the values...

>  I can not connect while the cranks are moving, I get the connection timeout message in this case, so the only way it connects is when I start the csc app before turning the sensor on

Well, that sounds like something related to the sensor?

Are you 100% sure that the old version of cscsensor actually connected fine? Because the connection code is identical - yes, there was that extra line for maxInterval, but that was in the wrong place so had zero effect on the connection itself.

nxdefiant, 2024-07-08:
> Are you 100% sure that the old version of cscsensor actually connected fine? 

Not quite fine, still got connection timeouts, but much less of them.

commit 8742e68e (2024-06-04): cscsensor 0.12: Fix regression reporting cadence (reported per second when should be per minute) (fix #3434)