espruino/BangleApps #3437: [runplus] Having both Run and Run+ is confusing

pavelmachek, 2024-05-28:
### Affected hardware version

Bangle 2

### Your firmware version

2vXX

### The bug

Could we merge functionality of Run and Run+, or perhaps just drop Run as Run+ is a superset? Having both is confusing, and it let me to debugging "why does not zooming in work for me". Well.. it did not as I had Run and feature is only available in Run+.

bobrippling, 2024-05-29:
Related: #2847 and this [commit comment thread](https://github.com/espruino/BangleApps/commit/625f31c8d44258bd977ce1a757a720e7da5f8ad0#commitcomment-114101694)

I think dropping Bangle 2 from `run` and Bangle 1 from `runplus` would solve the majority of the issue though

halemmerich, 2024-05-29:
I think we should keep the supports tag in metadata strictly for ability to work on the device. If we start deciding that on subjective factors we are going to have problems down the line. `run` is actually a nice example as `runplus` might deviate from the feature set further in the future. Having `run` available on Bangle 2 is useful, be it as the default app for running or even the lighter on RAM alternative if the user has other priorities and uses memory heavy stuff alongside `run`.

bobrippling, 2024-05-29:
Perhaps a suggested upgrade - if the user's selected BangleJS 2 as their device, we could have a metadata field which references the upgraded app, and we show that as a hint in the app loader

thyttan, 2024-06-08:
> Why did you get them mixed up? Maybe wording in readme could be clearer? Maybe `runplus` should have an altered icon? Suggestions and PRs welcome as always 😃

@pavelmachek did you have some feedback regarding this?

pavelmachek, 2024-06-08:
So what happened was that after various backups/restore cycles I was updating "run" application, wondering why I'm not getting new features from runplus.

thyttan, 2024-06-09:
Added hints in `run` app README and description: 13e25279b92c5395ce82add263da9459f2970e03 5b785bc5d14ea62f7e74c118daf1336c9dae4a8c

I think that's good enough for now.

closed 2024-06-09 by thyttan