espruino/BangleApps #3441: [mtnclock] Widgets disappear on Mountain Pass Clock

Bloodscars, 2024-06-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v21 107

### The bug

When using the Mountain Pass Clock app, the widget bar disappears  when the screen is locked. When unlocking the device, the widget bar shows up. After opening the launcher, then going back to the clock the widget bar disappears. When going to the app settings before going back to the clock, the Show Widgets button is no longer checked. The only consistent way I can see this bug happening is when Bluetooth connection is lost, though it happens even while still connected to Bluetooth.

### Installed apps

Clock: Mountain Pass Clock
Launcher: Icon Launcher & Pattern Launcher
Widgets:
- Message Widget
- Lock Widget
- Moon Phase
- Message Icons
- Simple Heart Rate Widget
- Weather
- Quiet Mode Schedule and Widget
- Bluetooth Widget with Notification
- A Battery Widget (with percentage)
- Widget Editor

thyttan, 2024-06-01:
Could you try installing stable fw 2v22 just in case that solves it?

Bloodscars, 2024-06-01:
> Could you try installing stable fw 2v22 just in case that solves it?

I uploaded 2v22 stable firmware and the Bluetooth connection issue causing the issue does not cause it anymore, however, the issue is still happening. The screen is locked, the widgets disappear, and then going to the Settings for Mountain Clock, the Show Widgets box is unchecked. I'm wondering if there is an issue with line 200 of the mtnclock app.js where it only draws the widgets, and then line 375 loads and then draws the widgets?

thyttan, 2024-06-01:
If your comfortable debugging this a PR with a bugfix is always very welcome! :)

If you connect the watch to the [Web IDE](https://espruino.github.io/EspruinoWebIDE/) you can see if you get any error messages when bugs are triggered. There are some pointers in [this tutorial](https://www.espruino.com/Bangle.js+Development).

Bloodscars, 2024-06-01:
I don't have a lot of experience with this, just some basic coding from classes. I was able to get the Web IDE Remote connection from the app onto my laptop and connected to Android. I am able to load the app from the Web IDE and I have Debug set to Both, but I cannot see any error logs or logs of any sort.

thyttan, 2024-06-01:
Thanks for testing that! So no outright errors it seems. But something about the logic like you say. Maybe you can use the Web IDE to tweak the loading and drawing of widgets and see if that fixes it?

Bloodscars, 2024-06-01:
I have done testing with the regular Mountain Clock and also with the extra line in the Web IDE, I think I have narrowed the issue down. When my app, Breezy Weather, updates it then breaks the widgets on the clock. In the Web IDE I get the following when I force a refresh with Breezy Weather (I changed the location to My Location instead of my city name)
```
Uncaught TypeError: Assignment to a constant
 at line 1 col 20
const _GB = global.GB;
                   ^
GB Unknown {
  "t": "weather",
  "temp": 298, "hi": 297, "lo": 289, "hum": 52, "rain": 0,
  "uv": 0, "code": 801,
  "txt": "Clouds",
  "wind": 7.42, "wdir": 90,
  "loc": "My Location"
 }
GB Unknown {
  "t": "weather",
  "temp": 298, "hi": 297, "lo": 289, "hum": 52, "rain": 0,
  "uv": 0, "code": 801,
  "txt": "Clouds",
  "wind": 7.42, "wdir": 90,
  "loc": "My Location"
 }
>
```
I have tested this with the Weather app, and with OWM app using my OpenWeather API Key, and also removed both apps and the same error happens when refreshing Breezy Weather with the Mountain Pass Clock. Not sure why it would disable the Show Widgets or why it crashes the widget bar. I think that the reason this happened previously when Bluetooth reconnected, it would refresh the weather data. And if that's the case, I think the reason why it happened "randomly", it was just when the weather would refresh in the background.

thyttan, 2024-06-01:
```
Uncaught TypeError: Assignment to a constant
 at line 1 col 20
const _GB = global.GB;
                   ^
```
What file is that from? Maybe you could try changing `const` to `let` there. Or the whole line to `if (!_GB) const _GB = global.GB;` or something like that.

Bloodscars, 2024-06-01:
It is from the mtnclock app.js file Line 343. I tried changing it to `let` instead of `const`, and also tried adding the line `if (!_GB) const _GB = global.GB`, as well as removing `const` and changing it to `let`, as it was giving me an error `Const declaration not directly within block` in the Web IDE.
However, after each of these changes I got the `GB Unknown` errors, but not the `Uncaught TypeError` error. The errors still persist even without the Show Widgets enabled, but visually there is no different between the error and the default clock.

thyttan, 2024-06-01:
Do you get those 'GB Unknown' errors also if you're using some other clock - maybe e.g. `antonclk`? Or are they specific to `mntnclk`?

I think what it's saying is "I don't have any instructions for what to do when t is passed the 'weather' value".

Before doing anything more, maybe give the 'Reinstall all apps' option on the app loader a try. If something is wrong in some boot file that could then fix it.

Bloodscars, 2024-06-01:
I tried this with Anton Clock and Weather Clock and the issue still persisted. I tried the Reinstall apps, as well as a factory reset (then adding Mountain Clock and Weather), and it is still having the `GB Unknown` errors when refreshing Breezy Weather.

thyttan, 2024-06-01:
Do you maybe not have the Android integration watch app, `android` installed? It should provide the instructions for the `"weather"` event ( https://www.espruino.com/Gadgetbridge#messages-sent-to-bangle-js-from-phone ).

Bloodscars, 2024-06-01:
I do have the Android app installed, the settings of the app are:

- `Connected Yes`
- `Find Phone >`
- `Keep Msgs []`
- `Overwrite GPS []`
- `Messages >`

thyttan, 2024-06-02:
OK, thanks! Well I'm out of ideas for now. I also don't have personal experience with weather syncing to lean on. Maybe I can try setting it up to see if I get the same as you. No ETA though 😛

If you work it out do tell! 🙂

thyttan, 2024-06-02:
I searched the repo for "GB Unknown" and got this: https://github.com/search?q=repo%3Aespruino%2FBangleApps+%22GB+Unknown%22&type=code

So it tries to assign the "weather" handler but it doesn't exist - as I verify by looking at https://github.com/espruino/BangleApps/blob/2a0cc2d66c84d9b958dfde34f78ec2914ed40086/apps/android/boot.js#L27 and below (or searching the file for "weather"). I might be missing something but it seems to me that `android` currently has no way of interpreting weather messages from gadgetbridge. Maybe this is done somewhere else though - it surely must have worked for others before?

Does weather work at all for you?

thyttan, 2024-06-02:
Searching for 'weather _GB' gives this: https://github.com/search?q=repo%3Aespruino%2FBangleApps+weather+_GB&type=code

So it seems like the weather app's library is responsible for handling weather messages - don't know really how it works myself.

Bloodscars, 2024-06-02:
The Weather app does work for me, though I am using the Bangle.js Gadgetbridge app and not the regular Gadgetbridge app. This allows the weather to function through Breezy Weather. I just tried with the regular Gadgetbridge app, set up Breezy Weather to send weather to it, and the same issue happens.

closed 2024-09-30 by bobrippling