espruino/BangleApps #3444: [OpenWeatherMap] com.android.volley AuthFailureError

doma9on, 2024-06-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v21

### The bug

OpenWeatherMap force refresh no longer works as I am met with 'com.android.volley AuthFailureError'. It seems to have been happening since they upgraded to API v3, so I created a new API key and saved it but I still get the error. Nothing was changed on the Bangle which could indicate why it no longer works.

bobrippling, 2024-06-16:
Yes, this is a v3 issue, there was a PR (#3427), but it wasn't merged. I'm sure @devsnd wouldn't mind if you wanted to bring it forward

doma9on, 2024-07-17:
> Yes, this is a v3 issue, there was a PR (#3427), but it wasn't merged. I'm sure @devsnd wouldn't mind if you wanted to bring it forward

Hi, thanks for reply. Will the PR be merged? If possible I would like to be able to use v2.5 if it means I don't have to enter payment details. If not, is there any other option for free weather data like this on Bangle? :)

devsnd, 2024-07-18:
@doma9on I closed the PR because the 2.5 API was discontinued in June. You either need to find another weather provider or add your payment details. I chose the latter, as it's practically free to use with the bangle still.

> [...] One Call API 2.5 has been finally closed in June 2024. 
https://openweathermap.org/one-call-transfer

zwiglm, 2024-07-26:
Sorry for repeating here!

I also got a paid subscription. From eg Postman works fine. Within the app, I still get the error message "com.android.volley..." when trying to force-update the weather.
Any suggestions?

zwiglm, 2024-07-26:
Hmm. Sorry. Now working