espruino/BangleApps #3454: Swiper clock launch and Pattern launcher apps no longer working with latest firmware

Moini, 2024-06-14:
After updating to espruino_2v22_banglejs2.zip , which seems to introduce changes with how gestures are handled and prioritized,

https://github.com/espruino/BangleApps/tree/master/apps/swiperclocklaunch
and most likely
https://github.com/espruino/BangleApps/tree/master/apps/ptlaunch

do not work anymore - which also makes it impossible to leave https://github.com/espruino/BangleApps/tree/master/apps/configurable_clock (leading to https://forum.espruino.com/conversations/397044/ ...)

Using Daisy clockface, Bangle.js 2

gfwilliams, 2024-06-14:
Moving to BangleApps...

I'm not aware of any changes to how gestures are handled and prioritized in 2v22

Can you be more specific about how exactly the apps don't work? And are you 100% sure that if you downgrade to 2v21 they work again?

Moini, 2024-06-15:
I have downgraded to 1.21 already, because after I had written this, I wasn't able to even leave the Daisy clock anymore. 

Daisy works again, with 1.2 with both the pattern launcher for the quick menu, and the swiper for the huge one.

The 'Configurable analog clock' is still not allowing me to leave it - I have since uninstalled it (and miss having a working and nice looking analog option).

gfwilliams, 2024-06-15:
So if I install daisy clock and swiperclocklaunch I'll be able to recreate it?

Moini, 2024-06-17:
I hope so... thanks!

For the pattern launcher, I added a downswipe pattern that goes down all three cells in the middle of the screen.

gfwilliams, 2024-06-17:
>> So if I install daisy clock and swiperclocklaunch I'll be able to recreate it?
> I hope so... thanks!

Just tested, and it works fine.

So there's not much we can do if we can't reproduce.

At minimum it'd be good if you could install 2v22 so it happens again, connect with the Web IDE and reload the clock (~2 sec button press) and try swiping.

My guess is there'll be some error message shown on the screen that will help track it down - but it's probably completely unrelated to the clocks/swiperclocklaunch

Moini, 2024-07-06:
Hi! So, there wasn't any error message.

But I've just updated all apps to current status, and the firmware to the latest 2v23, and the mini-launcher and swiping to go to the big launcher from Daisy work again.

closed 2024-07-06 by Moini