espruino/BangleApps #3466: Gadgetbridge Ack/Result Android->Bangle

Chriz76, 2024-06-20:
When sending a broadcast e.g. uart.tx  from Automate is there a way to check if the command reached the espruino device? Or is it even possible to get the console output of the statement?

I need this to do retries but in some cases it would also be interesting to read the result.

Gadgetbridge supports this for other non espruino intents by sending a broadcast with the result of an action. So it should be not too hard to add this to gadgetbridge as an optional parameter: SendAck/SendResult. 

What do you think?

I once commited another snall change to gadgetbridge so I could give it a try.

thyttan, 2024-06-20:
I don't have familiarity with those standard responses you write about. But it's of course possible to code the watch side to send a message back to Gadgetbridge, e.g. via the Intents messages: https://www.espruino.com/Gadgetbridge#intents

Chriz76, 2024-06-20:
> @gfwilliams are you familiar with GB and so on?

Gordon coded the gadgetbridge espruino integration.

Chriz76, 2024-06-20:
> I don't have familiarity with those standard responses you write about. 

Gadgetbridge's bluetooth intent api is one example. You can trigger a connect via an intent and Gadgetbridge will send a broadcast when the connection was established:
https://gadgetbridge.org/internals/automations/intents/#bluetooth-connection-api

> But it's of course possible to code the watch side to send a message back to Gadgetbridge, 

I guess that gadgetbridge already receives the result of the uart.tx call. Gadgetbridge currently just doesn't broadcast it. It also doesn't broadcast when it failed to deliver the calls.

thyttan, 2024-06-20:
> I guess that gadgetbridge already receives the result of the uart.tx call. Gadgetbridge currently just doesn't broadcast it. It also doesn't broadcast when it failed to deliver the calls.

I haven't looked at the Gadgetbridge source to verify but I don't think so. Also guessing here, but I think uart.tx only sends instructions over to the watch. If you want a response I think you need to explicitly program that yourself - either in the uart.tx message or in a destination app on the watch.

thyttan, 2024-06-20:
I think here is where the broadcasted intent is received and handled for Bangle.js: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/1f1fc33e4388a3386dd08ed79f0ae4ff8139e8ec/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L295

Chriz76, 2024-06-21:
> I think here is where the broadcasted intent is received and handled for Bangle.js: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/1f1fc33e4388a3386dd08ed79f0ae4ff8139e8ec/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L295

I looked at this code and assumed that sending the commands uses the existing ble connection that has a back channel:

https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/2cb5844020a4df0e7eb29180b32a8d93e76fbdd8/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L162

And it is handled in the same way on the bangle and in android like the web ide. So the result may already arrive in gadgetbridge.

gfwilliams, 2024-06-21:
That last link is to ~2 year old code btw.

We do currently send a local intent out when we get data back, but we don't want to broadcast it because it could contain sensitive info: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L1224-L1227

But realistically knowing *what* data you want to receive after sending a command is the problem. The UART is just a channel, and there's no 'framing' - data is sent and received but there's no way to say 'this data is the response to that command'.

As @thyttan says, it's best just to have the Bangle broadcast an intent back: https://www.espruino.com/Gadgetbridge#intents

At least then you know that your command has not only been sent, but has executed correctly too

Chriz76, 2024-06-21:
Thank all of you. Sending back the intent from my own code in bangle makes most sense. 

So currently I see no reason to keep this issue open. Thanks again!

closed 2024-06-21 by Chriz76