espruino/BangleApps #3476: [Settings] [Request] Reset Factory Settings

d3nd3, 2024-06-26:
### Affected hardware version

Bangle 1, Bangle 2

### Your firmware version

latest

### The bug

I think the reset factory settings feature under Utils is too easily pressed.
I once reset entire watch when thought i pressed turn off. Because turn off has same , are you sure dialog, it seemed to me the same when I was too fast.  Or if you accidently touch wrong and don't read.

I propose a double confirmation for factory reset. Where you say yes/no 2 times over.  It could also help to move the `turn off` feature to top of util list, then less chance for accident.

bobrippling, 2024-07-03:
Interesting point - @gfwilliams what do you think?

gfwilliams, 2024-07-04:
I have actually done this by accident once, so yes, I totally agree!

Personally I think the order should be swapped too, so 'turn off' is at the bottom, and then you can just scroll right down and press it.

I also think this tends to happen because when showing a menu, if you're scrolled such that a few pixels of the next menu items is showing at the bottom and you tap at the bottom then that menu item is selected. Perhaps we should modify `E.showMenu` to treat taps at the bottom as being on the last menu item if only a few pixels of the bottom menu item is showing (so users couldn't realistically have see what it was?)

thyttan, 2024-07-08:
https://github.com/espruino/Espruino/commit/ad25468f82bb7928f868168b04347dadad5a8eee

> Perhaps we should modify `E.showMenu` to treat taps at the bottom as being on the last menu item if only a few pixels of the bottom menu item is showing (so users couldn't realistically have see what it was?)

Done here: https://github.com/espruino/Espruino/commit/ad25468f82bb7928f868168b04347dadad5a8eee

commit 422a3492 (2024-07-10): 0.74: Add extra layer of checks before allowing a factory reset (fix #3476)
closed 2024-07-10 by gfwilliams