espruino/BangleApps #3481: [app/pebblepp] "Function "drawFilledImage" not found!"

pinq-, 2024-06-30:
### Affected hardware version

Bangle 2

### Your firmware version

2v23

### The bug

With the new update, I don't see any of the logos and getting error:
Function "drawFilledImage" not found!
 at line 5 col 13 in pebblepp.app.js
clock_info).drawFilledImage(info.img,midx-24,y,{scal

thyttan, 2024-06-30:
Did you do anything else beside updating the firmware between `pebblepp` working normally and it giving you that error?

Before doing anything else please do a backup of the watch and keep it around. We may ask you to share it for debugging - only if your comfortable of course.

After that I propose:
- Can you double check that you have the clock info module installed still?
- Can you connect with the Web IDE to see if the filesystem seems to have been somewhat corrupted?
- You can try the "Reinstall apps" button under the "More..." tab of the app loader.

I'm currently downgrading from cutting edge 2v23.8 to stable 2v23 to see if I can reproduce. On 2v23.8 it worked fine.

pinq-, 2024-07-02:
I updated pebblepp first and then firmare from 21 to 23.
How to I do a backup? Just copy all the files with the ide?

thyttan, 2024-07-02:
> How to I do a backup? Just copy all the files with the ide?

There's a backup option on the 'More...' tab of the app loader. Easiest to use that.

pinq-, 2024-07-02:
Found the solution: Had to update "Clock Info Module" app and then it stared to work. Thanks!

closed 2024-07-02 by pinq-