espruino/BangleApps #3484: [Sleep Quiet] Can't edit any setting

Tsht, 2024-07-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v23

### The bug

When I click on the app I have three settings (Quiet Switch, quiet mode, exit), however only Exit works. Clicking on the others only vibrates the watch, the setting won't change.
I should be able to edit the setting.
Just click on the watch button, then Sleep Quiet and try to change a setting.
I have SleepLog installed and working.

### Installed apps

Launcher (v0.21)
Bluetooth Widget (v0.09)
Lock Widget (v0.08)
Battery Level Widget (v0.12)
Messages (v0.62)
Weather (v0.26)
Alarms Widget (v0.02)
Pedometer widget (v0.22)
About (v0.15)
Message Icons (v0.07)
Anton Clock Plus (v0.11)
Message Widget (v0.06)
Message List (v0.05)
Simple Timer (v0.09)
Sleep Log (v0.18)
Scheduler (v0.26)
Notifications (default) (v0.14)
Android Integration (v0.36)
Settings (v0.72)
Alarms & Timers (v0.48)
Bootloader (v0.62)
Sleep Quiet (activate quiet mode when asleep) (v0.01)

bobrippling, 2024-07-02:
If you connect [the IDE](https://www.espruino.com/ide/), do you notice any errors emitted when you change those settings? @storm64 @Ishidres tagging with you two being the main devs

storm64, 2024-07-02:
I'd like to hand this over to @jla-42, as he is the creator of this app. From the bug description the sleeplog interface does not seem to be the problem.

jla-42, 2024-07-02:
will look into it.

jla-42, 2024-07-02:
> If you connect [the IDE](https://www.espruino.com/ide/), do you notice any errors emitted when you change those settings? 

the printout from the IDE would be really helpful. 
Thanks!

jla-42, 2024-07-02:
Hello @Tsht, I could reproduce the issue and (hopefully) fix it. 
I created a pull request, which is waiting for a review before it can be merged:  #3486 
If you would like to test it right away, you can do so in my fork: https://jla-42.github.io/BangleApps/?q=asleephttps://jla-42.github.io/BangleApps/?q=asleep

Thank you for finding the defect!

Best wishes
J

Tsht, 2024-07-02:
I think the file is created with v0.02, I saw this in the IDE (quietSwitch.json). It's empty ({}) but I can't still edit settings... I have no specific output on the IDE.

jla-42, 2024-07-02:
@Tsht >> please run this command in the ide, to erase the broken file, after that, the version 0.02 will during the first run create a new one populated with default values. hope that helps. 
require("Storage").erase("quietSwitch.json")

Tsht, 2024-07-02:
It's working after erasing the file, thanks ! ^^

jla-42, 2024-07-03:
Cool, thank you for testing! So we can now close this ticket.

bobrippling, 2024-07-04:
Thanks for the fix @jla-42!

closed 2024-07-04 by bobrippling

jla-42, 2024-07-04:
@bobrippling >> thank you for verification and closing!