espruino/BangleApps #3488: [Ratchet Launcher] High Memory Usage

jedthe3rd, 2024-07-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v22

### The bug

1. Install enough apps to have 31% ram usage
2. Install Ratchet Launcher
3. Launch it
4. Observe LOW_MEMORY, MEMORY error
I don't seem to have this issue with the Tag launcher or the default launcher.

### Installed apps

weather (0.26), weatherClock (0.06), 2047pp (0.04), diceroll (0.02), boot (0.62), setting (0.72), messageicons (0.07), messagegui (0.79), widmessages (0.06), messages (0.62), android (0.36), notify (0.14), hrm (0.13), health (0.30), widhrm (0.06), activityreminder (0.12), heart (0.09), widbt (0.09), sched (0.26), widalarm (0.02), widlock (0.08), recorder (0.43), locale (0.19), files (0.08), gpsinfo (0.13), authentiwatch (0.07), wid_a_battery_widget (0.04), cprassist (0.02), batchart (0.14), calculator (0.07), calendar (0.18), hidcam (0.04), chronowid (0.06), waypoints (0.04), gpsnav (0.09), counter (0.04), speedalt (0.14), gpstime (0.05), widhwt (0.02), myprofile (0.02), ha (0.12), hassio (0.02), info (0.03), medicalinfo (0.02), widmp (0.09), kbmulti (0.09), widpedom (0.22), mylocation (0.11), powermanager (0.11), runplus (0.27), sensible (0.07), sleeplog (0.18), smpltmr (0.09), sleeplogalarm (0.06), widsleepstatus (0.02), speedo (0.06), slevel (0.04), widtemp (0.02), thermom (0.07), torch (0.11), twenties (0.04), wid_edit (0.05), btadv (0.03), wohrm (0.10), multidice (1.30), magnav (0.07), barometer (0.04), altimeter (0.04), about (0.15), agpsdata (0.06), widram (0.03), alarm (0.48), clkinfosunrise (0.05), spotrem (0.11), quicklaunch (0.15)

jedthe3rd, 2024-07-03:
Just another note. I recently reset my watch because I was having all sorts of stability issues with it. So I installed all my apps 1 by 1 and I landed on the issues coming from the rachet launcher. Ever sense I stopped using it I have had no issues and my battery life is significantly longer (went from 2-3 days to 1-1.5 weeks) with the same HRM settings and I have more apps installed than I did before.

bobrippling, 2024-07-04:
This is mostly the result of a lot of bootapps being loaded, it could be that ratchet launcher is the final straw, or it's the largest of the bunch with the images it has loaded. Having said that, optimisations to drop its memory use are of course welcome

jedthe3rd, 2024-07-04:
> This is mostly the result of a lot of bootapps being loaded, it could be that ratchet launcher is the final straw, or it's the largest of the bunch with the images it has loaded. Having said that, optimisations to drop its memory use are of course welcome

I will do some more testing to see if I can get an understanding of the ram usage of just the launcher app. It's just weird it has so much more ram usage when compared to the launcher app it's based off of.

bobrippling, 2024-07-05:
Thanks - yeah that's quite odd, @gfwilliams do you know any good tricks for analysing memory usage as well as the [`process.memory(1)`](https://www.espruino.com/ReferenceBANGLEJS2#l_process_memory) call?