espruino/BangleApps #3492: [apploader] Can't connect to Bangle.js on Ubuntu

mkrl, 2024-07-04:
### Affected hardware version

Bangle 2

### Your firmware version

2v23

### The bug

When connecting to Bangle.js 2 from the App Loader and Espruino Web IDE I get an immidiate connection drop with a toast popping up saying
```
<TOAST>[error] Device connection failed, 
```
and no further info is provided like usual. 
Reading the logs in devtools (attached below) reveals the problem is most likely related to
```
GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`
```

### More context

While this bug occurs on a Ubuntu derivative (`Linux username 5.15.0-107-generic #117~20.04.1-Ubuntu SMP Tue Apr 30 10:35:57 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`) in Google Chrome `126.0.6478.61`, the connection is just fine while using Windows 11 on the same machine (both App Loader and Web IDE work).

I used to be able to do this on the same distro until I installed a firmware update to 2v23 (I am not entirely sure what was the previous version, but I think it was probably 2v17).

Shortly after the update I got a similar error to https://github.com/espruino/BangleApps/issues/631, which was fixed by resetting the Bangle. After that I was not able to connect using Chrome in Ubuntu.

### Devtools logs

<details>
  <summary>App Loader</summary>

  ```
=============================================
comms.js:3 Type 'Puck.debug=3' for full BLE debug info
comms.js:4 =============================================
espruinotools.js:5031 GET window.localStorage = null
espruinotools.js:54 Initialising Notifications
espruinotools.js:54 Initialising Status
espruinotools.js:54 Initialising Utils
espruinotools.js:54 Initialising Config
espruinotools.js:54 Initialising Serial
espruinotools.js:54 Initialising CodeWriter
espruinotools.js:54 Initialising Modules
espruinotools.js:54 Initialising Env
espruinotools.js:54 Initialising Compiler
espruinotools.js:54 Initialising Assembler
espruinotools.js:54 Initialising GetGitHub
espruinotools.js:54 Initialising Minify
espruinotools.js:54 Initialising CoreModules
espruinotools.js:54 Initialising Pretokenise
espruinotools.js:54 Initialising SaveOnSend
espruinotools.js:54 Initialising SetTime
pwa.js:5 👍 beforeinstallprompt BeforeInstallPromptEvent {isTrusted: true, platforms: Array(1), userChoice: Promise, type: 'beforeinstallprompt', target: Window, …}
Puck.debug=3
3
loader.js:190 Chosen device BANGLEJS2
puck.js:212 Web Bluetooth is experimental on this platform. See https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md
puck.js:409 <BLE> Device Name:       Bangle.js d2bc
puck.js:409 <BLE> Device ID:         bp93cz5jt+0LdCn7PfA33Q==
puck.js:409 <BLE> Connected
puck.js:409 <BLE> Disconnected (gattserverdisconnected)
ui.js:86 <TOAST>[error] Device connection failed, 
puck.js:409 <BLE> ERROR: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`
  ```

</details>

<details>
[... 83 lines cut ...]
index.js:96731 [notify_error] Connection Failed.
(anonymous) @ index.js:96731
error @ index.js:98122
(anonymous) @ index.js:104680
(anonymous) @ index.js:99648
closeSerial @ index.js:100660
(anonymous) @ index.js:100599
index.js:96721 >>> Connection Failed.
index.js:96721 BT> ERROR: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
  ```

</details>

### Steps attempted to fix

1. Multiple restarts and resets
2. Restarting the bluetooth service with `/etc/init.d/bluetooth restart`
3. Setting different combinations of "Programmable / BLE / HID"
4. Trying the connection in incognito mode to make sure no browser extensions mess it up

Let me know if I can provide more info to help! Thank you.

### Installed apps

boot (0.62), daisy (0.11), antonclk (0.11), widlock (0.08), widid (0.03), welcome (0.14), calculator (0.07), widbt (0.09), dvdbounce (0.01), warsawjs (undefined), widbat (0.12), pomodo (0.02), calendar (0.18), linuxclock (0.06), widalarm (0.02), flipper (undefined), hidmsicswipe (0.01), launch (0.21), sched (0.26), mylocation (0.11), notify (0.14), clock_info (0.12), setting (0.72), alarm (0.48), health (0.30), pebble (0.11), impwclock (0.07), slevel (0.04), sched (0.26), files (0.08), qrcode (0.06), about (0.15), recorder (0.43), run (0.19)

bobrippling, 2024-07-04:
Could be related to #2913

mkrl, 2024-07-04:
That might just be it @bobrippling, thank you!

I'll give it a try, I think I might have done a kernel update recently while not connecting Bangle for a long time before.
However, I still have some suspicions because I was able to connect before a firmware update (in order to do the update in the first place) and then not able to connect directly after updating.

thyttan, 2024-07-04:
Leaving some links in case they'd help: 
https://www.espruino.com/Troubleshooting+Bangle.js
https://www.espruino.com/Firmware+Update#nrf52
https://www.espruino.com/Troubleshooting+BLE

closed 2024-10-21 by gfwilliams